openrazer/openrazer #437: install driver on ubuntu 17.10, not working, dependency problems

rumplstielz, 2017-11-08:
Trying to install openrazer-meta, not working:

kurt@Yak:~$ sudo apt install openrazer-meta
[sudo] Passwort für kurt: 
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 openrazer-meta : Hängt ab von: openrazer-daemon soll aber nicht installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
kurt@Yak:~$ sudo apt install openrazer-daemon
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 openrazer-daemon : Hängt ab von: python3-evdev soll aber nicht installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
kurt@Yak:~$ sudo apt install python3-evdev
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 python3-evdev : Hängt ab von: python3 (< 3.6) aber 3.6.3-0ubuntu2 soll installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
kurt@Yak:~$ sudo apt install python3
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
python3 ist schon die neueste Version (3.6.3-0ubuntu2).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 3 nicht aktualisiert.
kurt@Yak:~$

z3ntu, 2017-11-08:
Please post the output of
`lsb_release -a`
and of
`cat /etc/apt/sources.list.d/*razer*`

rumplstielz, 2017-11-08:
kurt@Yak:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful
kurt@Yak:~$ cat /etc/apt/sources.list.d/*razer*
deb http://ppa.launchpad.net/openrazer/stable/ubuntu artful main
# deb-src http://ppa.launchpad.net/openrazer/stable/ubuntu artful main
deb http://ppa.launchpad.net/openrazer/stable/ubuntu artful main
# deb-src http://ppa.launchpad.net/openrazer/stable/ubuntu artful main
# deb http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04/ / # Bei Aktualisierung zu artful deaktiviert
deb http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04/ /
# deb http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04/ / # Bei Aktualisierung zu artful deaktiviert
deb http://ppa.launchpad.net/terrz/razerutils/ubuntu artful main # Bei Aktualisierung zu artful deaktiviert
# deb-src http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty main
deb http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty main
# deb-src http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty main
deb http://ppa.launchpad.net/terrz/razerutils/ubuntu artful main # Bei Aktualisierung zu artful deaktiviert
# deb-src http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty main

z3ntu, 2017-11-09:
Could you please try `apt update` and `apt install openrazer-meta` again?

rumplstielz, 2017-11-11:
Great job! Now it works! Thank you.

closed 2017-11-11 by rumplstielz

z3ntu, 2017-11-11:
You can also do `sudo rm /etc/apt/sources.list.d/terrz-ubuntu-razerutils*` to get rid of our old PPA (which caused your problem but we solved it server-side by deleting the old packages from that ppa)