openrazer/openrazer #438: failed launch RazerGenie

bcmmbc122333, 2017-11-09:
everytime I try to launch razer commander or razergenie they dont work. This is the error given from razergenie. I am currently using manjaro so that could be causing the error, but I thought I would check here first.

● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-11-08 18:18:14 CST; 182ms ago
     Docs: man:openrazer-daemon(8)
  Process: 1039 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 1039 (code=exited, status=1/FAILURE)

Nov 08 18:18:13 Brennen-PC systemd[760]: Starting Daemon to manage razer devices in userspace...
Nov 08 18:18:14 Brennen-PC systemd[760]: Started Daemon to manage razer devices in userspace.
Nov 08 18:18:14 Brennen-PC openrazer-daemon[1039]: 2017-11-08 18:18:14 | razer                          | INFO     | Initialising Daemon (v2.0.0). Pid: 1039
Nov 08 18:18:14 Brennen-PC openrazer-daemon[1039]: 2017-11-08 18:18:14 | razer                          | CRITICAL | User is not a member of the plugdev group
Nov 08 18:18:14 Brennen-PC systemd[760]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Nov 08 18:18:14 Brennen-PC systemd[760]: openrazer-daemon.service: Failed with result 'exit-code'.

z3ntu, 2017-11-09:
Your user is not a member of the plugdev group. To fix it, please run `sudo gpasswd -a <YOUR-USER> plugdev`.

bcmmbc122333, 2017-11-09:
now its launching but no devices are being detected by daemon. I am using manjaro with a deathadder chroma.

z3ntu, 2017-11-09:
Can you please run `sudo dkms install razer_chroma_driver/1.0.0` and post the output?

bcmmbc122333, 2017-11-09:
Creating symlink /var/lib/dkms/razer_chroma_driver/1.0.0/source ->
                 /usr/src/razer_chroma_driver-1.0.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.9.60-1-MANJARO cannot be found at
/usr/lib/modules/4.9.60-1-MANJARO/build or /usr/lib/modules/4.9.60-1-MANJARO/source.

z3ntu, 2017-11-09:
That means you have to install the kernel headers for your running kernel. (which seems to be `linux49-headers` on manjaro)

z3ntu, 2017-11-09:
Great! And next time please take a look at our [troubleshooting page](https://github.com/openrazer/openrazer/wiki/Troubleshooting) 😉

closed 2017-11-09 by z3ntu