openrazer/openrazer #442: No instructions for building

jcadduono, 2017-11-10:
Hi, I've looked all over the repository and wiki but haven't found any instructions for building.
Seeing the CMakeLists.txt I assumed cmake . && make && make install would be sufficient, however it seems the environment might require some variables set to work with kernel headers?
(my linux headers are properly installed and other projects using dkms build just fine)

```
jc@draenor:~/build/openrazer$ cmake .
-- The C compiler identification is GNU 7.2.1
-- The CXX compiler identification is GNU 7.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jc/build/openrazer
jc@draenor:~/build/openrazer$ make
Scanning dependencies of target FakeTarget
[ 16%] Building C object CMakeFiles/FakeTarget.dir/driver/razermouse_driver.c.o
/home/jc/build/openrazer/driver/razermouse_driver.c:22:10: fatal error: linux/slab.h: No such file or directory
 #include <linux/slab.h>
          ^~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/FakeTarget.dir/build.make:62: recipe for target 'CMakeFiles/FakeTarget.dir/driver/razermouse_driver.c.o' failed
make[2]: *** [CMakeFiles/FakeTarget.dir/driver/razermouse_driver.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/FakeTarget.dir/all' failed
make[1]: *** [CMakeFiles/FakeTarget.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

z3ntu, 2017-11-11:
The CMake file is just for code completion with CLion (an IDE).
The `Makefile` in the root of the repo is the main makefile which does everything.
You can look into it what targets exist but for installing the components of the project, see [here](https://github.com/openrazer/OBS-packaging/blob/master/PKGBUILD#L24). 
For just compiling the driver, you can just run `make` in the root of the repo and the driver will be compiled.

jcadduono, 2017-11-11:
ooooooo ok all I had to do was reset the repo and stop using the cmake command lol :(

closed 2017-11-11 by z3ntu