openrazer/openrazer #443: GNOME terminal slow to open after installing drivers in Ubuntu 16.04

ryanmmhmm, 2017-11-13:
Looking for an explanation or some understanding why this is happening.

It's only slow to open when I use the keyboard shortcut.

Thanks,
rm

z3ntu, 2017-11-13:
Have you tried uninstall the driver, rebooting and checking if our driver caused the issue?

ryanmmhmm, 2017-11-14:
Yes, but found that even after the uninstall the window is still slow to open.

The window was only slow to open once the driver was installed.

z3ntu, 2017-11-14:
Could you please rephrase your last comment? The two paragraphs are conflicting in their meaning.

KazWolfe, 2017-11-14:
Uninstall DBus, as it breaks *literally everything*.

    sudo apt remove dbus-user-session

Restart X and you'll be back online.

ryanmmhmm, 2017-11-14:
@KazWolfe It's not a problem with Chrome, but my GNOME terminal for the command line.

@z3ntu  I'll give a more thorough description of the problem to clear that up.

1. Keyboard and terminal were working fine.  Opening GNOME terminal window with keyboard shortcut CTRL + ALT + T would work instantly.

2. Installed open razer to use polychromatic add on.  CTRL + ALT + T now takes 10-20 seconds for the window to open.  Clicking on the terminal icon in my launcher bar opens it instantly.  Both are equivalent actions as far as I'm aware so I'm unsure why there's a difference.

3. Uninstalled the drivers for open razer and purged the repository.  Restarted the system.  CTRL + ALT + T still takes 10-20 seconds to open a window.

I hope that's more clear.

Cheers,
rm

z3ntu, 2017-11-14:
As the problem still exists after you have uninstalled drivers, it's not caused by our drivers... Except you can prove us otherwise.

KazWolfe, 2017-11-14:
@ryanmmhmm `dbus-user-session` breaks a lot of things. That includes Chrome and keyboard shortcuts.

Try it and then tell us if it doesn't work.

ryanmmhmm, 2017-11-14:
@KazWolfe @z3ntu - I removed `dbus-user-session` as suggested and restarted the system.  It's working normally again.

Is there any way for me to install open razer without installing dbus as a dependency?  Removing dbus removed all open razer and polychromatic...

KazWolfe, 2017-11-14:
A release that should patch this issue (v2.1) will be coming to the stable branch and PPA soon™.

In the meantime, you can use the daily PPA or manually download one of the prerelease DEBs which do not have a dependency on `dbus-user-session`.

ryanmmhmm, 2017-11-14:
@KazWolfe @z3ntu - thank you.  I installed from the daily ppa and the problem no longer occurs.

Cheers,
:beers:

closed 2017-11-14 by ryanmmhmm