openrazer/openrazer #444: Please Support CentOS (Version 7)

RisingPhorce, 2017-11-14:
Please consider supporting the distro CentOS 7.
Thanks!

z3ntu, 2017-11-14:
CentOS doesn't contain Python3 so we can't run our daemon on it. Duplicate of #248, #363 and #408

closed 2017-11-14 by z3ntu

philippeback, 2017-11-25:
With software collections python3 is fully supported on CentOS7

moozhub, 2017-11-26:
I built packages for CentOS 7 forever ago... just enable the EPEL repository and install openrazer-meta from https://github.com/moozhub/yum-repo-mooz ... I haven't had a single issue with it. I haven't really found a reason to build the latest version yet, but if someone wishes me to do so I don't have a problem doing it..

philippeback, 2017-11-26:
I forked that repo since I wasn't able to get the src rpm from it. Added the needed stuff as a local repo to be able to do a build from the sources.

There is still an issue which is that keys are not registering on my CentOS7.3.

Investigating.

moozhub, 2017-11-26:
As per the issue you opened, I built the repodata so you can pull the SRPMs down if you so desire.

philippeback, 2017-11-26:
Cool. Thx.