openrazer/openrazer #446: Daemon won't start

lucaszanella, 2017-11-17:
```
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 168, in <module>
    run()
  File "/usr/bin/openrazer-daemon", line 163, in run
    logger=logger)
TypeError: __init__() got an unexpected keyword argument 'chdir'
```

Did

`    print(args.run_dir)`

and got

`/run/user/1000`

lucaszanella, 2017-11-17:
If I take 

`                       foreground=args.foreground,`
and
`                       chdir=args.run_dir,`

out the daemon runs, but I get no devices found (which might be a different problem). Anyways, I compiled everything myself and installed. Do I need to do something more?

lucaszanella, 2017-11-17:
update: after reboot it worked

z3ntu, 2017-11-17:
What distro are you using?

lucaszanella, 2017-11-17:
Ubuntu 17.04.03

-------- Original Message --------

z3ntu, 2017-11-17:
So it works now or do you still need help?

lucaszanella, 2017-11-17:
it works but I think it did only because I restarted. Maybe someone will have this bug too.

-------- Original Message --------

z3ntu, 2017-11-17:
So we can close this?

lucaszanella, 2017-11-18:
yes

closed 2017-11-18 by z3ntu

JamieEi, 2018-03-29:
> I mean in the repository you have installed the openrazer packages from.

No, I get openrazer from
http://download.opensuse.org/repositories/hardware:/razer/Debian_Sid/.
python3-daemonize
comes from the Debian repos. Sid (unstable) has a recent version with the
missing __init__ flags but Stretch (stable) does not, including backports.

z3ntu, 2018-03-29:
You wrote in the last comment that you used http://download.opensuse.org/repositories/hardware:/razer/Debian_Sid/

JamieEi, 2018-03-29:
> Why did you use the Debian_Sid repo then instead of the Debian_9.0 repo if
> you use Debian 9?

Ah sorry, I must have grabbed the wrong command! Let me try again.