openrazer/openrazer #453: Process 'razer_mount razermouse 0003:1532:002F.0004' failed with exit code 1

shrugal, 2017-11-21:
Devices: Imperator 2012 and DeathStalker Essential
OS: Fedora 27 x64

I installed the driver and software via the rpm repo, but Polychromatic says it couldn't recognize the device and I get the error message above in on boot. I know the keyboard is not supported yet, but I would like to use the driver for the mouse at least.

Output of `journalctl -xb | grep -i razer`:

```
Nov 21 22:45:19 niko-pc kernel: usb 3-3: Product: Razer DeathStalker Essential
Nov 21 22:45:19 niko-pc kernel: usb 3-3: Manufacturer: Razer
Nov 21 22:45:19 niko-pc kernel: input: Razer Razer DeathStalker Essential as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0119.0001/input/input2
Nov 21 22:45:19 niko-pc kernel: hid-generic 0003:1532:0119.0001: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer DeathStalker Essential] on usb-0000:00:14.0-3/input0
Nov 21 22:45:19 niko-pc kernel: input: Razer Razer DeathStalker Essential as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/0003:1532:0119.0002/input/input3
Nov 21 22:45:19 niko-pc kernel: hid-generic 0003:1532:0119.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer DeathStalker Essential] on usb-0000:00:14.0-3/input1
Nov 21 22:45:19 niko-pc kernel: input: Razer Razer DeathStalker Essential as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:1532:0119.0003/input/input4
Nov 21 22:45:19 niko-pc kernel: hid-generic 0003:1532:0119.0003: input,hidraw2: USB HID v1.11 Mouse [Razer Razer DeathStalker Essential] on usb-0000:00:14.0-3/input2
Nov 21 22:45:19 niko-pc kernel: usb 1-1.5: Product: Razer Imperator
Nov 21 22:45:19 niko-pc kernel: usb 1-1.5: Manufacturer: Razer
Nov 21 22:45:19 niko-pc kernel: input: Razer Razer Imperator as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0/0003:1532:002F.0004/input/input5
Nov 21 22:45:19 niko-pc kernel: hid-generic 0003:1532:002F.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Imperator] on usb-0000:00:1a.0-1.5/input0
Nov 21 22:45:19 niko-pc kernel: input: Razer Razer Imperator as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.1/0003:1532:002F.0005/input/input6
Nov 21 22:45:19 niko-pc kernel: hid-generic 0003:1532:002F.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Imperator] on usb-0000:00:1a.0-1.5/input1
Nov 21 21:45:21 niko-pc razer_mount[526]: Driver razermouse
Nov 21 21:45:21 niko-pc razer_mount[528]: Device_ID 0003:1532:002F.0004
Nov 21 21:45:21 niko-pc razer_mount[529]: Modprobing razermouse
Nov 21 21:45:21 niko-pc systemd-udevd[511]: Process 'razer_mount razermouse 0003:1532:002F.0004' failed with exit code 1.
Nov 21 21:45:21 niko-pc razer_mount[527]: Driver razermouse
Nov 21 21:45:21 niko-pc razer_mount[534]: Device_ID 0003:1532:002F.0005
Nov 21 21:45:21 niko-pc razer_mount[537]: Modprobing razermouse
Nov 21 21:45:21 niko-pc systemd-udevd[503]: Process 'razer_mount razermouse 0003:1532:002F.0005' failed with exit code 1.
Nov 21 21:45:23 niko-pc systemd-logind[929]: Watching system buttons on /dev/input/event2 (Razer Razer DeathStalker Essential)
Nov 21 21:45:23 niko-pc systemd-logind[929]: Watching system buttons on /dev/input/event3 (Razer Razer DeathStalker Essential)
Nov 21 21:45:23 niko-pc systemd-logind[929]: Watching system buttons on /dev/input/event6 (Razer Razer Imperator)
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) config/udev: Adding input device Razer Razer DeathStalker Essential (/dev/input/event2)
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: Applying InputClass "evdev keyboard catchall"
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: Applying InputClass "libinput keyboard catchall"
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: Applying InputClass "system-keyboard"
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) Using input driver 'libinput' for 'Razer Razer DeathStalker Essential'
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: always reports core events
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event2  - (II) Razer Razer DeathStalker Essential: (II) is tagged by udev as: Keyboard
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event2  - (II) Razer Razer DeathStalker Essential: (II) device is a keyboard
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event2  - (II) Razer Razer DeathStalker Essential: (II) device removed
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) XINPUT: Adding extended input device "Razer Razer DeathStalker Essential" (type: KEYBOARD, id 8)
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event2  - (II) Razer Razer DeathStalker Essential: (II) is tagged by udev as: Keyboard
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event2  - (II) Razer Razer DeathStalker Essential: (II) device is a keyboard
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) config/udev: Adding input device Razer Razer DeathStalker Essential (/dev/input/event3)
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: Applying InputClass "evdev keyboard catchall"
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: Applying InputClass "libinput keyboard catchall"
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: Applying InputClass "system-keyboard"
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) Using input driver 'libinput' for 'Razer Razer DeathStalker Essential'
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: always reports core events
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event3  - (II) Razer Razer DeathStalker Essential: (II) is tagged by udev as: Keyboard
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event3  - (II) Razer Razer DeathStalker Essential: (II) device is a keyboard
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event3  - (II) Razer Razer DeathStalker Essential: (II) device removed
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) libinput: Razer Razer DeathStalker Essential: needs a virtual subdevice
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) XINPUT: Adding extended input device "Razer Razer DeathStalker Essential" (type: MOUSE, id 9)
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: (accel) selected scheme none/0
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: (accel) acceleration factor: 2.000
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: (accel) acceleration threshold: 4
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event3  - (II) Razer Razer DeathStalker Essential: (II) is tagged by udev as: Keyboard
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event3  - (II) Razer Razer DeathStalker Essential: (II) device is a keyboard
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) config/udev: Adding input device Razer Razer DeathStalker Essential (/dev/input/event4)
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: Applying InputClass "evdev pointer catchall"
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: Applying InputClass "libinput pointer catchall"
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) Using input driver 'libinput' for 'Razer Razer DeathStalker Essential'
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (**) Razer Razer DeathStalker Essential: always reports core events
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event4  - (II) Razer Razer DeathStalker Essential: (II) is tagged by udev as: Mouse
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event4  - (II) Razer Razer DeathStalker Essential: (II) device is a pointer
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) event4  - (II) Razer Razer DeathStalker Essential: (II) device removed
Nov 21 21:45:38 niko-pc /usr/libexec/gdm-x-session[1608]: (II) XINPUT: Adding extended input device "Razer Razer DeathStalker Essential" (type: MOUSE, id 10)
[... cut ...]
Nov 21 21:45:43 niko-pc openrazer-daemon[2182]: 2017-11-21 21:45:43 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
Nov 21 21:45:43 niko-pc openrazer-daemon[2182]: 2017-11-21 21:45:43 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
Nov 21 21:45:43 niko-pc openrazer-daemon[2182]: 2017-11-21 21:45:43 | razer                          | INFO     | Serving DBus
Nov 21 21:45:43 niko-pc openrazer-daemon[2182]: 2017-11-21 21:45:43 | razer                          | DEBUG    | DBus called get_serial_list
```

hyph3n, 2017-11-22:
Device: Razer Naga Hex v2
OS: Fedora 26

The daemon (with Polychromatic) has been working perfectly since I got the mouse a couple of months ago. After a dnf update the other day, Polychromatic is now saying the device isn't recognized by the daemon. Group membership has not changed since things were working.

```
dmesg | grep -i razer

[    1.024656] usb 3-2: Product: Razer Naga Hex V2
[    1.024657] usb 3-2: Manufacturer: Razer
[    1.035219] input: Razer Razer Naga Hex V2 as /devices/pci0000:00/0000:00:07.1/0000:11:00.3/usb3/3-2/3-2:1.0/0003:1532:0050.0001/input/input2
[    1.035501] hid-generic 0003:1532:0050.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Naga Hex V2] on usb-0000:11:00.3-2/input0
[    1.041982] input: Razer Razer Naga Hex V2 as /devices/pci0000:00/0000:00:07.1/0000:11:00.3/usb3/3-2/3-2:1.1/0003:1532:0050.0002/input/input3
[    1.093547] hid-generic 0003:1532:0050.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Naga Hex V2] on usb-0000:11:00.3-2/input1
[    1.098221] input: Razer Razer Naga Hex V2 as /devices/pci0000:00/0000:00:07.1/0000:11:00.3/usb3/3-2/3-2:1.2/0003:1532:0050.0003/input/input4
[    1.150593] hid-generic 0003:1532:0050.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Naga Hex V2] on usb-0000:11:00.3-2/input2
```

```
sudo journalctl -b | grep -i razer_mount

Nov 21 08:46:47  razer_mount[1112]: Driver razermouse
Nov 21 08:46:47  razer_mount[1113]: Driver razermouse
Nov 21 08:46:47  razer_mount[1115]: Driver razermouse
Nov 21 08:46:47  razer_mount[1116]: Device_ID 0003:1532:0050.0002
Nov 21 08:46:47  razer_mount[1117]: Device_ID 0003:1532:0050.0001
Nov 21 08:46:47  razer_mount[1128]: Modprobing razermouse
Nov 21 08:46:47  razer_mount[1127]: Modprobing razermouse
Nov 21 08:46:47  razer_mount[1125]: Device_ID 0003:1532:0050.0003
Nov 21 08:46:47  systemd-udevd[1039]: Process 'razer_mount razermouse 0003:1532:0050.0001' failed with exit code 1.
Nov 21 08:46:47  razer_mount[1141]: Modprobing razermouse
Nov 21 08:46:47  systemd-udevd[1045]: Process 'razer_mount razermouse 0003:1532:0050.0002' failed with exit code 1.
Nov 21 08:46:47  systemd-udevd[1067]: Process 'razer_mount razermouse 0003:1532:0050.0003' failed with exit code 1.
```

```
cat ~/.local/share/openrazer/logs/razer.log

017-11-21 08:47:04 | razer                          | INFO     | Initialising Daemon (v2.1.1). Pid: 3684
2017-11-21 08:47:04 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-11-21 08:47:04 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
[... cut ...]
2017-11-21 08:47:04 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2017-11-21 08:47:04 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2017-11-21 08:47:04 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2017-11-21 08:47:04 | razer                          | INFO     | Serving DBus
2017-11-21 08:47:04 | razer                          | DEBUG    | DBus called get_serial_list
```

z3ntu, 2017-11-22:
Could one of you (whoever is faster) please run `sudo modprobe razermouse` and `sudo dkms status` and post the output here?

shrugal, 2017-11-22:
I got a kernel update yesterday evening from 4.13.12 to 4.13.13 and now it seems to work!

Output from modprobe is empty and from dkms it's

```
openrazer-driver, 1.0.0, 4.13.13-300.fc27.x86_64, x86_64: installed
```

I'll leave the issue open in case @hyph3n still has problems.

z3ntu, 2017-11-22:
Weird... But good it's working again :)
I'll wait for @hyph3n to answer

hyph3n, 2017-11-22:
Here's what I'm seeing

```
sudo modprobe razermouse

modprobe: FATAL: Module razermouse not found in directory /lib/modules/4.13.13-200.fc26.x86_64
```

```
sudo dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/razer_chroma_driver/1.0.0/source/dkms.conf does not exist.
```

Also perhaps worth pointing out:

```
uname -a

Linux 4.13.13-200.fc26.x86_64 #1 SMP Wed Nov 15 15:46:36 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

Thanks!

z3ntu, 2017-11-22:
```
sudo rm -rf /var/lib/dkms/razer_chroma_driver/
sudo dkms install openrazer-driver/1.0.0
```
that should hopefully fix the issue.

hyph3n, 2017-11-22:
Would that require a reboot? I've run the commands and restarted the daemon, but still facing the same situation.

z3ntu, 2017-11-22:
Was the dkms install command successful? If yes, replugging your usb devices should do the trick.

hyph3n, 2017-11-22:
Disconnected and reconnected the mouse, and then restarted the daemon. All good now! Thanks!

z3ntu, 2017-11-22:
Great. Sorry for the breakage.

closed 2017-11-22 by z3ntu