openrazer/openrazer #454: daemon not starting

Kliesen, 2017-11-22:
I can't get the daemon to start, I tried updating, recompiling(from branch "feature_abyssus1800"), reinstalling and rebooting. I get the following error:
`Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 168, in <module>
    run()
  File "/usr/bin/openrazer-daemon", line 163, in run
    logger=logger)
TypeError: __init__() got an unexpected keyword argument 'foreground'`

sometimes the word "foreground" changes to "chdir", using ubuntu gnome 17.04

z3ntu, 2017-11-22:
I think you are using an old version of python3-daemonize. Could you please try installing python3-daemonize from the PPA and trying again?

Kliesen, 2017-11-22:
I updated from daily ppa and now it's working, also the abyssus 1800 support works pretty well on polychromatic, thanks.

z3ntu, 2017-11-22:
Great 🙂

closed 2017-11-22 by z3ntu

Ayak973, 2018-02-15:
Hello, just for your information, with Debian (4.9.0-4-amd64 SMP Debian 4.9.65-3+deb9u1 (2017-12-23) x86_64) on a Macbook Pro 17" mid 2011, the Opensuse repo didn't work even following the tutorial, I had the same error ([...] unexpected keyword argument 'foreground').
I'd have to install python3-daemonize version 2.4.7-1 from the official Debian Sid repository, then install openrazer-meta directly from the Ubuntu xenial repo.
Regards

z3ntu, 2018-02-15:
@Ayak973 You say it didn't work on openSUSE, which version of openSUSE was this?
And can you tell me what steps you took on Debian (again, which version?) when it didn't work?

z3ntu, 2018-02-18:
I just tested Debian 9.3 in a VM and everything works as expected. 
[image]

Ayak973, 2018-02-18:
Well, the kernel version of my installation differs from yours (4.9.0-5, 2018-01-04 vs 4.9.0-4, 2017-12-23) maybe a recent update has fixed the issue.
Sorry for the inconvenience...

z3ntu, 2018-02-18:
The problem described in the first comment of this issue has nothing to do with the kernel. Well if it works now, it's good.