openrazer/openrazer #455: Razer Naga Chroma Daemon Crash

izatrix, 2017-11-23:
Plychromatic was complaining of a max dpi crash. I removed poly and openrazer demon and repositories and added daily ppa's for both but the deamon is still crashing.  I thought in the daily build the dpi check was fixed. Did I miss something here? I'd like to use my side buttons in games.

Oh this is ubuntu 16.04.03 or .04 LTS my kernal is: 4.10.0-40-generic x64
Polychromatic is complaining about daemon issues

`polychromatic-controller -v
Verbose enabled
Set scale to: 1.0
Initialised Window.
Version 0.3.11.2-stable
Loading daemon...
Gathering devices...
 -- Found: Razer Naga Chroma
Loading application profiles...

Set active device: Razer Naga Chroma
Device supports 'lighting_logo_brightness'
Device supports 'lighting_scroll_brightness'
There was a problem communicating with the daemon for this device.
Exception: org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 654, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 246, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: maxDPI is not a valid method of interface razer.device.dpi
`

`polychromatic-controller --print-device-info
===============================================================================
Please include this data when reporting device-specific issues on GitHub.
===============================================================================

Connected Devices:          
[0] - Razer Naga Chroma (Serial: PM1605H00801237)

Enter choice: 0

Selected: Razer Naga Chroma

-------------------------------------------------------------------------------
```
# Device Info
Name: Razer Naga Chroma
Type: mouse
 
# Device Capabilities
{'lighting': True, 'lighting_scroll_blinking': False, 'lighting_breath_random': True, 'lighting_scroll_breath_single': True, 'macro_logic': True, 'firmware_version': True, 'lighting_scroll_static': True, 'lighting_logo_static': True, 'lighting_logo_active': False, 'poll_rate': True, 'lighting_starlight_single': False, 'lighting_breath_single': True, 'lighting_none': True, 'lighting_backlight_active': False, 'dpi': True, 'lighting_spectrum': True, 'lighting_logo_brightness': True, 'serial': True, 'lighting_breath_triple': False, 'lighting_logo_blinking': False, 'lighting_backlight': False, 'lighting_logo_breath_single': True, 'lighting_logo_none': True, 'lighting_starlight_dual': False, 'lighting_scroll_breath_random': True, 'lighting_scroll_none': True, 'lighting_logo': True, 'lighting_logo_pulsate': False, 'lighting_scroll_pulsate': False, 'lighting_reactive': True, 'lighting_logo_breath_random': True, 'name': True, 'lighting_breath_dual': True, 'lighting_ripple': False, 'lighting_pulsate': False, 'lighting_scroll_spectrum': True, 'lighting_logo_breath_dual': True, 'lighting_ripple_random': False, 'brightness': False, 'lighting_scroll_breath_dual': True, 'lighting_scroll_active': False, 'lighting_wave': False, 'lighting_starlight_random': False, 'lighting_scroll_reactive': True, 'lighting_static': True, 'lighting_logo_reactive': True, 'lighting_scroll': True, 'lighting_led_single': False, 'type': True, 'lighting_scroll_brightness': True, 'lighting_led_matrix': True, 'lighting_logo_spectrum': True}
```
-------------------------------------------------------------------------------
`
`rm ~/.local/share/openrazer/logs/razer.log
razer-daemon -r
razer-daemon: command not found
`

I'm not sure what else to post any suggestions would be greatly appreciated!

z3ntu, 2017-11-23:
Can you please try to apply the patch https://github.com/z3ntu/openrazer/commit/3d0e39f387e9371398475d254c356055b3a4477c to the file `/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/mouse.py` (the part with "python3.6" could be different depending on the distro)

harm, 2017-11-23:
I have exactly the same issue. 
Polychromatic compains: 

There was a problem communicating with the daemon for this device.
Exception: org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 654, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 246, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: maxDPI is not a valid method of interface razer.device.dpi

z3ntu, 2017-11-23:
@harm Please try the patch mentioned in my last comment.

izatrix, 2017-11-24:
That did fix the dpi issue however I'm still getting daemon crasses it just crashed once but I didn't catch the information however in: ~/.local/share/openrazer/logs/razer.log I'm getting this issue also.

`Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 10
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 11
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 11
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 9
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 10
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
`

izatrix, 2017-11-24:
That did fix the dpi issue however I'm still getting daemon crasses it just crashed once but I didn't catch the information however in: ~/.local/share/openrazer/logs/razer.log I'm getting this issue also.

`Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 10
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 11
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 11
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-11-23 19:27:39 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 9
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 13
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 10
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 12
2017-11-23 19:27:40 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
`
I'm also seeing this crash from polychromatic I'm not sure if it's related at all or if I should post it in the polychromatic github.
[image]

zethra, 2017-12-26:
I'm having the same issue

stillenacht288, 2017-12-28:
I'm having similar issues, running Ubuntu 16.04. My log has the same errors as izatrix.

z3ntu, 2017-12-28:
Please update to the latest version v2.2.0 and try again.