openrazer/openrazer #456: ConsoleKit support

ivazqueznet, 2017-11-23:
Attached is a script that generates a udev rules file that gives the logged-in user access to Razer devices without needing to belong to plugdev. It should work on any modern distro that uses udev and ConsoleKit. Pass it the directory containing the driver headers and it outputs the rules file to stdout.

    bash genrazerudev.sh openrazer/driver > 60-razer-daemon.rules
    sudo cp 60-razer-daemon.rules /etc/udev/rules.d

[genrazerudev.sh.txt](https://github.com/openrazer/openrazer/files/1499953/genrazerudev.sh.txt)

z3ntu, 2017-11-23:
You tested it and it works?

ivazqueznet, 2017-11-23:
Sure did.

```
$ getfacl /dev/input/mouse2 
getfacl: Removing leading '/' from absolute path names
# file: dev/input/mouse2
# owner: root
# group: plugdev
user::rw-
user:ignacio:rw-
group::rw-
mask::rw-
other::---
```

z3ntu, 2017-11-23:
Are you sure this is also applying on the sysfs files the driver is creating (so e.g. in `/sys/bus/hid/drivers/razerkbd/0*/`)?

ivazqueznet, 2017-11-23:
It doesn't appear to be affecting sysfs. Let me do a bit more work on it...

ivazqueznet, 2017-11-23:
sysfs doesn't support ACLs, so this can't affect that. Never mind...

closed 2017-11-23 by ivazqueznet

z3ntu, 2017-11-23:
Yes, that's what my own research resulted in :(

z3ntu, 2017-11-23:
Thanks for trying though 😃

ivazqueznet, 2017-11-23:
It could work if the driver created /dev entries instead, but more work.