openrazer/openrazer #459: Kernel issues with RBS 2017

samr28, 2017-11-26:
Running the latest stable Debian Kernel with KDE on my RBS 2017. Here are the issues that I'm running into:

1. When the device is connected to AC power, it goes into hibernate mode. This only happens if the device has gone into hibernate mode since being powered on. For example, everything works fine when I first boot the device but if I close the lid and open it again, then I will run into this issue.

2. Speakers seem to pop when adjusting volume. They were working fine but after updating this package they seem to be making weird popping noises. This may be a hardware issue but I'm not quite sure. I can record some of the noises if that would be helpful.

3. When the laptop lid is closed, the screen does not turn off. If I open and close the screen again then it will turn off. As far as I can tell, it always turns off when closing the second time. It works fine the first time but after the first hibernate (similar to the first issue) I need to open and close the lid a second time for the screen to turn off.

I'm pretty new to Linux in general so this may be the wrong place to list these bugs. If so, please let me know where I should post em.

z3ntu, 2017-11-26:
Unfortunately all the issues you listed are not caused by this project and are completely out of the scope of this project. We just provide kernel drivers + daemon for changing the lights on Razer hardware.

closed 2017-11-26 by z3ntu

samr28, 2017-11-26:
ok. I thought this might be an issue with the kernel drivers. That's what someone on the Debian IRC said. Where else should I post this? The only other place I can think of is the Debian forum. Any suggestions would be appreciated!

z3ntu, 2017-11-26:
I don't really know where... The audio stuff is maybe caused by pulseaudio? The others seem to be some software issue in some library somewhere ^^ 
And about what's our fault: You can try to uninstall openrazer (for testing only of course 😉 ) and test if any of this still happens and if it is, it's not our fault and we can't do anything to fix it.

KazWolfe, 2017-11-26:
This definitely is not related to Openrazer.

The popping noise [is known](https://insider.razerzone.com/index.php?threads/razer-blade-pro-light-related-speaker-noise.27734/) and we don't know how to fix it. Seems like a grounding issue or something, but nobody's sure how to fix it yet (that I know of). Read that thread, and try turning off the lights to see if the sound dies. The more data we have, the better.

Close state issues/hibernation are [also known](https://insider.razerzone.com/index.php?threads/laptop-lid-close-state-and-ac-power-connection-problems.27661/). The close state [seems to be fixable](https://insider.razerzone.com/index.php?threads/rbs-with-i7-8th-gen-ubuntu-17-10.27310/#post-389186), add the following to your `GRUB_CMDLINE_LINUX_DEFAULT` in `/etc/default/grub`:

    button.lid_init_state=open

Run `sudo update-grub` and restart. Things should suspend properly, or at least they do on my Blade Pro. This *may* also resolve your hibernate issue, let us know.

samr28, 2017-11-26:
Thank you so much! That seemed to resolve both of the power-related issues! I'm glad that I'm not the only one with the speaker popping noises! Mine seems to be a bit different than the ones listed on the forum you linked. I'll make a little post there to provide some more info. Thanks again!!