openrazer/openrazer #461: Unable to lock on the pidfile.

ceiphr, 2017-11-29:
Hi,
About two days ago polychromatic started telling me that it was unable to recognize any devices on my linux machine (Gnome Ubuntu 16.04 on Razer Blade Stealth 2016). I reestablished the PPAs for both openrazer and polychromatic and reinstalled both. I also checked the open razer logs and didn't see any errors within the past few days. When I run openrazer-daemon -Fv the output is Unable to lock on the pidfile. I'm unsure what to do or if this issue has been resolved. Hopefully someone can either help with the issue or point me in the right direction. Thanks.

z3ntu, 2017-12-02:
Please try running `pkill -e openrazer && openrazer-daemon -Fv`

ceiphr, 2017-12-06:
It's working again. Thank you!

closed 2017-12-06 by ceiphr

Sicks3c, 2020-12-02:
Hello 

I'm having the same issue at `Arch Linux`

## Output of pkill -e openrazer && openrazer-daemon -Fv

```
openrazer-daemo killed (pid 3096)
Starting daemon.
2020-12-02 06:40:06 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 11538
2020-12-02 06:40:06 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlade15Advanced2020 ---------------- (1532:0253)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Base ---------------------- (1532:0246)
2020-12-02 06:40:06 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019StudioEdition ------------- (1532:024D)
[... cut ...]
2020-12-02 06:40:06 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2020-12-02 06:40:06 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2020-12-02 06:40:06 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2020-12-02 06:40:06 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2020-12-02 06:40:06 | razer                          | INFO     | Serving DBus

```

It stuck on the terminal like that

Cheers

z3ntu, 2020-12-02:
@Sicks3c: That means that the daemon is running normally. Please open a new ticket if you experience any problems and write them there.