openrazer/openrazer #462: Fedora RBS not detected

kprinssu, 2017-11-29:
I am on Fedora 27 and the Razer Blade Stealth (1532:0220 ) is not detected:

```
$ openrazer-daemon -Fv
Starting daemon.
2017-11-29 11:59:06 | razer                          | INFO     | Initialising Daemon (v2.1.1). Pid: 5899
2017-11-29 11:59:06 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerMambaTE ---------------------------- (1532:0046)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerNagaChroma ------------------------- (1532:0053)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaverChroma -------------------- (1532:0207)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2011 ------------------------- (1532:0013)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2013 ------------------------- (1532:0039)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2017-11-29 11:59:06 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
[... cut ...]
2017-11-29 11:59:06 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2017-11-29 11:59:06 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2017-11-29 11:59:06 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2017-11-29 11:59:06 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2017-11-29 11:59:06 | razer                          | INFO     | Serving DBus
```

z3ntu, 2017-11-29:
Did you install the correct kernel headers (`kernel-devel` probably)? Please give us the output of `sudo dkms status && sudo dkms install openrazer-driver/1.0.0`

kprinssu, 2017-11-29:
I did expecility install the kernel headers. Running the command:

```
# sudo dkms status && sudo dkms install openrazer-driver/1.0.0
openrazer-driver, 1.0.0, 4.13.15-300.fc27.x86_64, x86_64: installed
Module openrazer-driver/1.0.0 already installed on kernel 4.13.15-300.fc27.x86_64/x86_64
```

Showing that kernel-devel is installed:
````
# yum list kernel-devel
Last metadata expiration check: 0:25:22 ago on Wed 29 Nov 2017 11:57:25 AM EST.
Installed Packages
kernel-devel.x86_64                                                                                    4.13.15-300.fc27                                                                                    @updates
````

z3ntu, 2017-11-29:
That's good. Have you tried rebooting the device?

kprinssu, 2017-11-29:
Do you mind if I get the command to do so?

z3ntu, 2017-11-29:
`ls -al /sys/bus/hid/drivers/razerkbd/0*/` ?

kprinssu, 2017-11-29:
Here's the output:

```
 ls -al /sys/bus/hid/drivers/razerkbd/0*/
ls: cannot access '/sys/bus/hid/drivers/razerkbd/0*/': No such file or directory
```

z3ntu, 2017-11-29:
I'm guessing rebooting will fix it but if you won't reboot I can't really help you..

kprinssu, 2017-11-29:
That indeed fix the issue.

closed 2017-11-29 by kprinssu