openrazer/openrazer #464: Razer Cynosa Chroma Pro

nfugal, 2017-12-02:
Here's what I've got. I'll add screen shots of the options a bit later

` nfugal  ~  Documents  lsusb -d 1532: -v

Bus 001 Device 016: ID 1532:022c Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x022c 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
 nfugal  ~  Documents  ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 280 Dec  1 06:35 .
drwxr-xr-x 4 root root 740 Dec  1 06:35 ..
lrwxrwxrwx 1 root root  10 Nov 28 22:37 usb-Logitech_USB_Receiver-if02-event-kbd -> ../event23
lrwxrwxrwx 1 root root  10 Nov 28 22:37 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event22
lrwxrwxrwx 1 root root   9 Nov 28 22:37 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse6
lrwxrwxrwx 1 root root  10 Dec  1 06:35 usb-Razer_Razer_Cynosa_Chroma_Pro-event-kbd -> ../event18
lrwxrwxrwx 1 root root  10 Dec  1 06:35 usb-Razer_Razer_Cynosa_Chroma_Pro-if01-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10 Dec  1 06:35 usb-Razer_Razer_Cynosa_Chroma_Pro-if02-event-mouse -> ../event20
lrwxrwxrwx 1 root root   9 Dec  1 06:35 usb-Razer_Razer_Cynosa_Chroma_Pro-if02-mouse -> ../mouse4
lrwxrwxrwx 1 root root  10 Nov 28 22:37 usb-Skullcandy__Inc._Skullcandy_GMX_Stereo_Transmitter-event-if03 -> ../event21
lrwxrwxrwx 1 root root  10 Nov 27 22:40 usb-SunplusIT_Inc_Integrated_Camera-event-if00 -> ../event17
lrwxrwxrwx 1 root root   9 Nov 27 22:40 usb-Wacom_Co._Ltd._Pen_and_multitouch_sensor-event-if00 -> ../event4
lrwxrwxrwx 1 root root   9 Nov 27 22:40 usb-Wacom_Co._Ltd._Pen_and_multitouch_sensor-if01-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 Nov 27 22:40 usb-Wacom_Co._Ltd._Pen_and_multitouch_sensor-if01-mouse -> ../mouse1
 nfugal  ~  Documents  
`

terrycain, 2017-12-02:
Screenshots from Razer synapse please

nfugal, 2017-12-02:
I'm working on getting a Windows VM setup so I can get them.

terrycain, 2017-12-02:
@nfugal 👍

nfugal, 2017-12-02:
[image]
[image]
[image]
[image]
[image]
[image]

nfugal, 2017-12-02:
It's the beta of Synapse v3

nfugal, 2017-12-02:
Are there other particular screens I should get?

terrycain, 2017-12-02:
They are fine. Can I get usb packet captures of the following (guide [here](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol))
* 1 capture of moving brightness up and down a few times
* 1 capture of game mode on / off
* 1 capture of each quick effect and all their settings
a text file to say what you did per capture is also helpful.

nfugal, 2017-12-02:
[Actions Taken.txt](https://github.com/openrazer/openrazer/files/1524270/Actions.Taken.txt)

[USB Captures.zip](https://github.com/openrazer/openrazer/files/1524271/USB.Captures.zip)

nfugal, 2017-12-02:
I'm happy to answer any questions, or to redo any of the captures as needed.

z3ntu, 2017-12-04:
Could you try to make a capture (like none->static->none) with Synapse version 2? Just for seeing if those are any different.

nfugal, 2017-12-04:
I haven't been able to interact with the keyboard in Synapse v2. When you plug the keyboard in it automatically starts an install on the 3 beta.

Vehyla, 2018-06-06:
Is any more work going into these drivers?  I've been trying to PR for the Cynosa Chroma and just changing the usb id from 022a to 022c.  Sadly it does not work.  Would love to use this keyboards features.

JoshMcCullough, 2018-06-14:
Hoping this would also work with the non-Pro version of this keyboard.

z3ntu, 2018-06-14:
Cynosa Chroma support is tracked in #586

aneergaard, 2018-10-03:
What is the status on the Pro version?

FuzzyGamers, 2018-10-06:
Bumping Aneergaard, is there any update?

aneergaard, 2019-05-05:
Any update?

You do great work, keep it up!

gabrielfnd, 2020-04-24:
Hey guys, any update?
I can provide help if needed.

nightsky30, 2020-04-26:
@gabrielfnd Not making any promises, but would you like to try this test [branch](https://github.com/nightsky30/openrazer/tree/feature_cynosa_chroma_pro)?  The Pro may be close enough to the Cynosa Chroma that this is all that is needed.

gabrielfnd, 2020-04-27:
@nightsky30 hello,
I successfully could build the package from the branch you provided. 
Yes, it is working. But I noticed that the "Reactive" and "Ripple" effects aren't working.
Great job nonetheless :smiley:

nightsky30, 2020-04-27:
@gabrielfnd Great!

 - Regarding the ripple effect, can you pull the most recent change and retest?  I found a typo that might make a difference.

 - Is the underglow working well with the Cynosa Chroma Pro?  I was hoping it might be similar to the Huntsman Elite.  It looked like the Huntsman Elite had extra rows in its lighting diagram in Synapse 3, and they added extra rows in https://github.com/openrazer/openrazer/commit/c8d944bd94ea1d1e74149d7367855b7fb6efb5d6 to enable the underglow.  With the Cynosa Chroma Pro there did not seem to be extra rows visible, so I left the standard matrix of [6, 22].

gabrielfnd, 2020-04-27:
@nightsky30, 

- Pulled the most recent change, but no luck with that effect yet
- Yes, the underglow is working great, as you can see [image]

nightsky30, 2020-04-27:
Hmmm...not sure about the ripple.  It was a direct copy/pasta from the Cynosa Chroma portion of keyboards.py, so I would think ripple should be working so long as the hardware supports what ripple_effect.py is doing.  Glad everything else is working :)

nightsky30, 2020-05-17:
@neergaard @FuzzyGamers @Vehyla @nfugal Would any of you care to test #1117 and the ripple effect? Seems ripple is the one outlying issue, and a change was made that might fix the problem.

PR #1117 (2021-02-27): Adds support for Cynosa Chroma Pro
Adds support for Cynosa Chroma Pro.  Fixes #464.

Reported working:
 - [x] Breath Single
 - [x] Breath Dual
 - [x] Breath Random
 - [x] Wave
 - [x] Reactive
 - [x] None
 - [x] Spectrum
 - [x] Static
 - [x] Brightness
 - [x] Underglow

Not added / tested (might need custom frames mode switch):
 - Ripple
 - Ripple Random

Tested by @gabrielfnd. Everything appears to be working minus the ripple effect.
closed 2021-02-27 by z3ntu