openrazer/openrazer #465: Multiple Keys using same keycodes across mouse and keyboard

DaveDesrochers, 2017-12-03:
So right now I only have openrazer installed, can't get the gui applications(razergenie, polychromatic) to do anything useful so right now none are installed. After switching the keyboard(Blackwidow ultimate 2013) into driver mode the M keys are recognized as 191-195, however, it also has mapped the additional keys from my mouse(Ouroboros) to 191-195. So I can't seem to be able to use the mouse keys separate from the keyboard M keys. Also for some reason, the right thumb rest button is mapped to keycode 16 which is bound to 7 on the keyboard.

terrycain, 2017-12-03:
... hmm, don't have ouroboros to test with, but you sure driver mode made keys appear with 191-195? (With keyboard thats expected)

DaveDesrochers, 2017-12-03:
Yea, I've been using a xev with awk filter I found on the arch wiki. Here's the output. The first command was M1-5. Then I re-ran it going through the extra keys on the mouse. This is with both set to driver mode. Without driver mode, it will only pick up the two buttons below the scroll wheel and assign them to 191 and 192 with one set of the thumb buttons non-functional.
[image]

jonnyasmar, 2017-12-12:
Having the same issue here, except my Ouroboros' side keys (the ones used to floor the mouse DPI) are being recognized as 8 and 9.

DaveDesrochers, 2017-12-16:
So after digging through the source code, I've found something that looks a little suspicious and might be causing these problems. At razermouse_driver.c:1936 in the razer_raw_event function it processes the 'DPI up' button in the exact same manner as razerkbd_driver.c:1610 processes the 'M1' key and appears to produce the same data byte array. I don't understand enough about writing Linux drivers to fix this or say its definitely the problem but I don't intend to give up.

tildearrow, 2017-12-16:
This is my fault.

Some months ago I sent a pull request that would enable the DPI buttons and scroll wheel tilting (#229). I decided to use the keycodes for F13-F16 (which the drivers already use for M1-M5) since I couldn't get the driver to send mouse scroll events or something (#201).

There are 2 (or maybe 3?) solutions if I'm sure:
- Do not enable driver mode on your mouse, unless you need to remap the DPI buttons (please confirm if this is possible).
- I can send another pull request to change the keycodes so they don't conflict.

jonnyasmar, 2017-12-16:
Second option sounds good to me! :)

DaveDesrochers, 2017-12-16:
In driver mode the dpi buttons are mapped to F13-F14 and do not change the dpi when pressed. Some of the small silver thumb buttons on the right get maped to F15-F16(the ouroburous doesn't have a tilting scroll wheel) while the thumb 'paddles' gets maped to keycodes 16-17 so they type 7 and 8 when pressed. In normal mode the small silver thumb buttons prevously mapped to F15-F16 on the right appear dead but the 'paddles' are still on 16-17 and the DPI buttons remain mapped to F13-F14 and change the dpi when pressed. 

First option doesn't really work. Also is the paddles registering under keycodes 16-17 a known issue? I can deal without them but having it documented might be helpful.

terrycain, 2017-12-16:
Yeah this is a big issue across nearly all mice with extra buttons and the gamepads like tartarus. And Then you need to deal with setting event values on the mouse that will be different from every keyboard value and any values decided to be set on the tartarus etc...

closed 2018-05-15 by stale[bot]

malsyned, 2020-05-08:
Is it possible to walk up the hierarchy from the keyboard interface to the USB device, then back down the mouse interface and emit a button click event for these higher mouse buttons? from BTN_SIDE up, X (specifically xorg-xf86-input-libinput) just sequentially assigns button numbers starting at 8. Presumably these could be mapped in games, by a daemon, etc.?

malsyned, 2020-05-08:
This is consistent with what the driver for my Logitech Performance MX does.