openrazer/openrazer #466: Razer Lachesis

joychan469, 2017-12-06:
lsusb -d 1532: -v
------------------------
```
Bus 009 Device 002: ID 1532:000c Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x000c 
  bcdDevice           21.00
  iManufacturer           1 Razer
  iProduct                2 Razer Lachesis
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      73
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      47
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

Bus 003 Device 002: ID 1532:021f Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x021f 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Ornata
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

ls -lah /dev/input/by-id/
------------------------------
```
total 0
drwxr-xr-x 2 root root 140 Dec  7 01:16 .
drwxr-xr-x 4 root root 460 Dec  7 01:16 ..
lrwxrwxrwx 1 root root   9 Dec  7 01:16 usb-Razer_Razer_Lachesis-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 Dec  7 01:16 usb-Razer_Razer_Lachesis-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Dec  7 01:16 usb-Razer_Razer_Lachesis-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Dec  7 01:16 usb-Razer_Razer_Ornata-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Dec  7 01:16 usb-Razer_Razer_Ornata-if01-event-kbd -> ../event7
```

Screenshots of all available options on Razer Synapse. (not applicable - Not available on Synapse. However, standalone razer software for the Lachesis is.) (Note: 1,2,4,6,5,7 have the exact same menu items. 8 and 9 has "Macro" select by default even though it doesn't exist in the dropdown list.)
------------------------------------------------------
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Product: Razer Lachesis
Model Number: RZ01-0017
Product Number: RZ01-00170100-R3M1

z3ntu, 2017-12-06:
Please make packet captures (see the wiki how to) of:
* "Scroll Button" on->off->on
* "Snake Logo" on->off->on
* Polling rate 125Hz -> 500Hz -> 1000Hz
* DPI 1000->2000->4000

joychan469, 2017-12-06:
[Razer-Lachesis-PCAP.zip](https://github.com/openrazer/openrazer/files/1536561/Razer-Lachesis-PCAP.zip)

z3ntu, 2017-12-06:
Thanks. Could you try to make a quick capture (doesn't matter what option) and explicitely save the file in the .pcapng format (instead of .pcap)? Also, what Wireshark version did you use?

joychan469, 2017-12-06:
I had trouble with Wireshark (using Windows 10 [not VM]), probably due to my inexperienced. So I captured directly from elevated CMD and running USBPcapCMD.exe directly, specifying the filter to monitor (after noting the number), then capturing the movements via this method. Adapted from: http://desowin.org/usbpcap/tour.html

joychan469, 2017-12-06:
Wireshark presents me with 9 USBPcap "channels" and only USBPcap1 has data. However, data shown from running USBPcapCMD.exe shows that my Lachesis is under USBPcap8. However, ending a capture on USBPcap8 on Wireshark shows a warning that says no packets are captured.

joychan469, 2017-12-06:
And to answer your question, I'm using Wireshark 2.4.3. Now returning to windows to make the quick capture you wanted in .pcapng

joychan469, 2017-12-06:
[Quick-random-capture.zip](https://github.com/openrazer/openrazer/files/1536634/Quick-random-capture.zip)

[image]

Weird. Now USBPcap8 returns some packets. Maybe its because i've first loaded USBPcapCMD.exe. Let me redo all the captures.

joychan469, 2017-12-06:
[PCAPNG.zip](https://github.com/openrazer/openrazer/files/1536688/PCAPNG.zip)

@z3ntu its done.

Astaoth, 2018-04-02:
Hello,
I'm also interrested by the lachesis support. Can we have some news about it ? :)
Cheers

dpalmigiano, 2018-05-22:
Yes, please do fill us in. How might we help?

2ivanpa, 2021-02-19:
@z3ntu @dpalmigiano 
Hi! As far as I guess issue hasn't been solved yet but was planned to a milestone 2.5 years ago.
So... are you so much behind the schedule or do you have updated planned delivery date?

z3ntu, 2021-02-27:
@2ivanpa The milestones have been misused as labels, I've fixed that now. 

Someone with the device will have to add support to openrazer if this is ever going to be supported. It's a really old device so it uses an old protocol unlike most other devices supported. But we do have a few in the tree, like the Orochi 2011 or the DeathAdder 3.5G, where the Lachesis probably uses a similar structure. 
This file https://github.com/mbuesch/razer/blob/master/librazer/hw_lachesis.c from a different project might also come in useful as reference.