openrazer/openrazer #469: Razer BlackWidow Ultimate 2016 - ripple effect and macros support

presianbg, 2017-12-07:
Hi,

Just go my Razer BlackWidow Ultimate 2016, but I'm not able to find a way to set my ripple effect nor the key macros recording.
```
 cat 0003\:1532\:0214.001B/device_type 
Razer BlackWidow Ultimate 2016

```

Any tips?

Thanks!

z3ntu, 2018-01-30:
Does ripple work now? So this issue is only about macros now?

presianbg, 2018-01-30:
I will check If the same ( high CPU usage) issue with ripple effect still exists.

z3ntu, 2018-01-30:
Yes, ripple is CPU intensive. Pull requests with a better version are welcome.

presianbg, 2018-02-01:
I will look at it, but I'm sure you did it right... just thought it's a bug. 
I used to believe that the logic is handled by device its self :(

closed 2018-02-01 by presianbg

z3ntu, 2018-02-01:
No, unfortunately ripple is not a hardware effect. 

Also Synapse v3 on Windows spams the USB devices all the times with packets what LEDs they should turn on compared to Synapse v2 where it mostly only sent something to the device when an effect changed.