openrazer/openrazer #470: Better reversing documentation

C0rn3j, 2017-12-09:
https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

>Now you are ready to start Wireshark, start it as root. 

No, bad! You should just add the user to the wireshark group(and relog) and use setcap if that is not enough.

    sudo gpasswd -a c0rn3j wireshark
    sudo setcap 'CAP_NET_RAW+eip CAP_NET_ADMIN+eip CAP_DAC_OVERRIDE+eip' /usr/bin/dumpcap

>Phase 4 (Command Decoding)

As a wireshark noob, this section was totally unhelpful.

Add instructions to add certain things in a wireshark column for easy viewing, and add a note how stupid wireshark views things. There actually is an example of this already - 

    00ff00000004030a0600ff00000000000000000000000000...
   00ff00000002030a01020000000000000000000000000000...

You can see the three dots at the end cut off part of the capture. The user can either click through each capture on wireshark and read it in bytes, or export it via tcpdump and possibly colorize it using my tool, as described here.

https://gitlab.com/C0rn3j/a4tech_bloody_p85_driver/#dev

I think adding all this info will make the guide way more newbie friendly

z3ntu, 2017-12-09:
I've deactivated the write protection on the wiki so you should be able to edit it. Thank you for improving the wiki 👍

C0rn3j, 2017-12-09:
I've begun editing the wiki, however it's written with a Razer keyboard as an example, and I do not have a Razer keyboard, much less any Razer device.

I need someone to take screenshot(s) of razer synapse app keyboard options.

z3ntu, 2017-12-09:
Looks much better already 😄

C0rn3j, 2017-12-09:
Thanks!

Also I'll need someone to provide a pcap file from Wireshark (any that has a few back-and-forths will do really) when debugging a Razer device - I want to test if my colorizing script will work with it.

z3ntu, 2017-12-09:
[abyssus1800.zip](https://github.com/openrazer/openrazer/files/1544724/abyssus1800.zip)
some caps from the abyssus 1800

C0rn3j, 2017-12-09:
Thanks, I've added BULK to the list of ignored strings in my script, now it seems to work ok with the razer captures.

[dpi change.pcapng.txt](https://github.com/openrazer/openrazer/files/1544741/dpi.change.pcapng.txt)

Could you try inputting the content of this file in a form on this site, submitting it and seeing if it looks correct? 
https://sc3.c0rn3j.com/a4/

Does it make sense to you/would be helpful when reversing Razer devices? (blue is command from HOST to device, green is from device to host)

---

I can see it still leaves things like this, (which I'll fix)
```
02ff 0000 0003 0401 4444 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
0x0040: 0000 0000 0000 0000 0000 0000 0000 0000
0x0050: 0000 0000 0000 0000 0600
```
in it, which it shouldn't (it works great on my A4tech devices which only have 0x000 to 0x030 captures, will look into making it more general)

z3ntu, 2017-12-09:
We already have a python script for parsing some commands from the packet captures

C0rn3j, 2017-12-09:
Well, I've added and modified practically everything I wanted to.

Would be cool if someone with more knowledge than me added some reference implementation as Phase 5.

Also adding at least a single Synapse screenshot would be great, since the example is focused on Synapse.

Could also be a good idea to mention your parsing script in there?

C0rn3j, 2018-04-09:
<activity>
The wiki page would still be missing reference implementation, screenshots and the parsing script.

C0rn3j, 2019-01-16:
(posting to not have this closed)
The wiki page would still be missing reference implementation, screenshots and the parsing script.