openrazer/openrazer #472: Razer Tartarus is represented as "UNKNOWN DEVICE"

gump74, 2017-12-09:
I have a Razer Tartarus and RazerGenie says it is an unknown device. I followed the install instructions and read through as many of other issues that I could but still genie is not recognizing it. I am only a basic Linux user but I have tried to go through as much troubleshooting as I know. I found this in my dmesg
`module verification failed: signature and/or required key missing`
But at [this](https://www.linuxquestions.org/questions/linux-virtualization-and-cloud-90/module-verification-failed-signature-and-or-required-key-missing-tainting-kernel-4175533442/) site they say it is not a big deal. I noticed that while it was plugged in my space bar does not work on the keyboard and when I use two fingers to scroll on a page it goes crazy, but only while the tartarus is plugged in. I did a lsusb -d [uid] -v and all that seems to be ok but I can attach it you like. That would seems to indicate that the driver is working right? modprobe does not have any feedback. The other logs that I looked at do not have any specific entires that seem relevant. I am using Ubuntu 17.10. I'm not sure what else I should check. Can anyone help? Thanks in advance.

z3ntu, 2017-12-09:
You have to disable secure boot or sign the kernel modules after each update.

gump74, 2017-12-09:
I read several pages about disabling secure boot and did not know what to do. One page said I needed to go into bios. I read issue 86 and I did not understand how to sign the modules. How do I do sign the modules?

Nice response time BTW.

z3ntu, 2017-12-09:
As I have never used Secure Boot, I don't know how to sign the modules.
To disable secure boot you need to reboot into the UEFI (press some key while booting to get in there, depends on the computer) and in some menu there you should be able to disable secure boot. 

p.s.: we have a troubleshooting page where this issue is listed 😉 https://github.com/openrazer/openrazer/wiki/Troubleshooting

gump74, 2017-12-09:
I'll try to disable it on boot up and see what that does.

gump74, 2017-12-09:
all of the references that I can find say that secure boot is tied to uefi. uefi happens when you select encrypt my hard drive durring install. I did not do this and I do not see a another way to do this. I do not know what to do.

C0rn3j, 2017-12-09:
>uefi happens when you select encrypt my hard drive durring install.

This makes no sense.

UEFI is the firmware of your motherboard. Traditionally you get into it by mashing F1/F2/DEL/F11/F6 right after POST. Varies wildly by the manufacturer.

You can also use systemd to reboot into UEFI

    sudo systemctl reboot --firmware-setup

gump74, 2017-12-09:
I do not understand this feature, I was just reciting what I read. I got into bios and checked the setting for secure boot. It was already off.

z3ntu, 2017-12-09:
Can you please run `sudo modprobe razerkraken && lsmod | grep razer` ?

gump74, 2017-12-09:
I have the output here:
```
mike@harvey:~$ sudo modprobe razerkraken && lsmod | grep razer
[sudo] password for mike: 
razerkraken            20480  0
hid                   118784  1 razerkraken
mike@harvey:~$ 
```

z3ntu, 2017-12-09:
Seems to work... Please post the output of `ls -al /sys/bus/hid/drivers/razerkbd/0*/` and 
`python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` and also attach the file `~/.local/share/openrazer/logs/razer.log`

gump74, 2017-12-09:
output 1
```
mike@harvey:~$ ls -al /sys/bus/hid/drivers/razerkbd/0*/
ls: cannot access '/sys/bus/hid/drivers/razerkbd/0*/': No such file or directory
mike@harvey:~$ sudo ls -al /sys/bus/hid/drivers/razerkbd/0*/
ls: cannot access '/sys/bus/hid/drivers/razerkbd/0*/': No such file or directory
mike@harvey:~$ sudo ls -al /sys/bus/hid/drivers/razerkbd/
ls: cannot access '/sys/bus/hid/drivers/razerkbd/': No such file or directory
mike@harvey:~$ sudo ls -al /sys/bus/hid/drivers/r*
total 0
drwxr-xr-x 2 root root    0 Dec  9 14:51 .
drwxr-xr-x 3 root root    0 Dec  9 14:45 ..
--w------- 1 root root 4096 Dec  9 14:51 bind
lrwxrwxrwx 1 root root    0 Dec  9 14:51 module -> ../../../../module/razerkraken
--w------- 1 root root 4096 Dec  9 14:51 new_id
--w------- 1 root root 4096 Dec  9 14:45 uevent
--w------- 1 root root 4096 Dec  9 14:51 unbind
mike@harvey:~$ 

```

output2

```
mike@harvey:~$ python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[]

```

[razer.log](https://github.com/openrazer/openrazer/files/1545139/razer.log)

gump74, 2017-12-09:
did I do output2 correctly?

z3ntu, 2017-12-09:
Weird...
`lsusb | grep 1532`?

and yeah, output2 is what I wanted to know

gump74, 2017-12-09:
I do apologize the razer was not plugged in because I cannot use my spacebar. I'll do it again

gump74, 2017-12-09:
output 2
```
mike@harvey:~$ python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[<RazerKeyboard PM1338103406181>]

```

output3
```
mike@harvey:~$ lsusb | grep 1532
Bus 003 Device 002: ID 1532:0201 Razer USA, Ltd 

```

z3ntu, 2017-12-09:
Everything's fine according to this. Is there another problem still?

gump74, 2017-12-09:
when I open razergenie the tartarus is displayed but is labeled as "unknown device"

z3ntu, 2017-12-09:
Please run `python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices[0].name)"`

gump74, 2017-12-09:
output
```
mike@harvey:~$ python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices[0].name)"
Unknown Device
```

razergenie was not open, do I need to open it?

z3ntu, 2017-12-09:
No they are 2 seperate applications. Now please run `cat /sys/bus/hid/drivers/razerkbd/0*/device_type`

gump74, 2017-12-09:
```
mike@harvey:~$ cat /sys/bus/hid/drivers/razerkbd/0*/device_type
Unknown Device

```

gump74, 2017-12-09:
thanks for all your time BTW

z3ntu, 2017-12-09:
Will be included in the next release. Thanks.

closed 2017-12-09 by z3ntu

gump74, 2017-12-09:
is my hardware not presently supported?

z3ntu, 2017-12-09:
It is, just the device name in the driver was missed.

gump74, 2017-12-09:
oh i see, how will I know when the next release is out? can I set something in apt to watch it?

gump74, 2017-12-09:
is it just the name? the razergenie cannot configure the device buttons just has a slider for the brightness.

z3ntu, 2017-12-10:
Button remapping isn't yet supported by the daemon.

commit 71f52c87 (2017-12-09): Add missing device_type to driver

Closes #472