openrazer/openrazer #474: Razer Abyssus 1800 (1532:0020) support

williamcanin, 2017-12-12:
Apparently, it does not seem to support my Razer Abyssus 1800 mouse. Is there any way to get support in the future? Thanks. :)

Info:

$ lsusb | grep Razer
Bus 002 Device 003: ID 1532:0020 Razer USA, Ltd

$ lsusb -d 1532:0020 -v
Bus 002 Device 003: ID 1532:0020 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0020 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Abyssus 1800
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      41
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

$ ls -lah /dev/input/by-id/
drwxr-xr-x 2 root root 160 dez 12 07:17 .
drwxr-xr-x 4 root root 480 dez 12 07:17 ..
lrwxrwxrwx 1 root root  10 dez 12 07:17 usb-Chicony_Electronics_Co.__Ltd._Sony_Visual_Communication_Camera_Sony_Visual_Communication_Camera-event-if00 -> ../event15
lrwxrwxrwx 1 root root   9 dez 12 07:17 usb-Razer_Razer_Abyssus_1800-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 dez 12 07:17 usb-Razer_Razer_Abyssus_1800-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 dez 12 07:17 usb-Razer_Razer_Abyssus_1800-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 dez 12 07:17 usb-USB_USB_Keyboard-event-if01 -> ../event7
lrwxrwxrwx 1 root root   9 dez 12 07:17 usb-USB_USB_Keyboard-event-kbd -> ../event6

z3ntu, 2017-12-12:
Duplicate of #365

closed 2017-12-12 by z3ntu

williamcanin, 2017-12-12:
Oh God, I should conduct a search before posting. I'm really sorry (pull my ear).