openrazer/openrazer #475: ArchLinux with GNOME (Wayland) : Doesn't work + Tried everything on "Troubleshooting" + Searched a lot

OverSurge, 2017-12-15:
Hello,
I can't manage to make the driver work properly.
I'm running ArchLinux with GNOME on Wayland, system is up to date.
I'm having errors at multiple places :

`[ 1933.439899] razerkbd: loading out-of-tree module taints kernel.` with `dmesg`

`sudo openrazer-daemon -Fv
Starting daemon.
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 123, in run_daemon
    test_dir=args.test_dir)
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/daemon.py", line 92, in __init__
    DBusService.__init__(self, self.BUS_NAME, '/org/razer')
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/dbus_services/service.py", line 83, in __init__
    bus_object = dbus.service.BusName(bus_name, bus=dbus.SessionBus())
  File "/usr/lib/python3.6/site-packages/dbus/_dbus.py", line 211, in __new__
    mainloop=mainloop)
  File "/usr/lib/python3.6/site-packages/dbus/_dbus.py", line 100, in __new__
    bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 122, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ExecFailed: /usr/bin/dbus-launch terminated abnormally with the following error: No protocol specified
Autolaunch error: X11 initialization failed.
` (Without sudo, it doesn't start)

sudo journalctl -b | grep -i razer_mount : 
`Dec 15 18:50:40 archlucas razer_mount[302]: Driver razerkbd
Dec 15 18:50:40 archlucas razer_mount[301]: Driver razermouse
Dec 15 18:50:40 archlucas razer_mount[299]: Driver razerkbd
Dec 15 18:50:40 archlucas razer_mount[300]: Driver razermouse
Dec 15 18:50:40 archlucas razer_mount[303]: Driver razermouse
Dec 15 18:50:40 archlucas razer_mount[304]: Device_ID 0003:1532:0203.0001
Dec 15 18:50:40 archlucas razer_mount[306]: Device_ID 0003:1532:0034.0005
Dec 15 18:50:40 archlucas razer_mount[305]: Device_ID 0003:1532:0203.0002
Dec 15 18:50:40 archlucas razer_mount[308]: Modprobing razerkbd
Dec 15 18:50:40 archlucas razer_mount[307]: Device_ID 0003:1532:0034.0006
Dec 15 18:50:40 archlucas razer_mount[310]: Modprobing razerkbd
Dec 15 18:50:40 archlucas razer_mount[311]: Device_ID 0003:1532:0034.0007
Dec 15 18:50:40 archlucas razer_mount[309]: Modprobing razermouse
Dec 15 18:50:40 archlucas razer_mount[314]: Modprobing razermouse
Dec 15 18:50:40 archlucas razer_mount[315]: Modprobing razermouse
Dec 15 18:50:40 archlucas systemd-udevd[231]: Process 'razer_mount razerkbd 0003:1532:0203.0002' failed with exit code 1.
Dec 15 18:50:40 archlucas systemd-udevd[235]: Process 'razer_mount razerkbd 0003:1532:0203.0001' failed with exit code 1.
Dec 15 18:50:40 archlucas systemd-udevd[244]: Process 'razer_mount razermouse 0003:1532:0034.0005' failed with exit code 1.
Dec 15 18:50:40 archlucas systemd-udevd[234]: Process 'razer_mount razermouse 0003:1532:0034.0007' failed with exit code 1.
Dec 15 18:50:40 archlucas systemd-udevd[238]: Process 'razer_mount razermouse 0003:1532:0034.0006' failed with exit code 1.
Dec 15 18:50:40 archlucas razer_mount[322]: Driver razerkbd
Dec 15 18:50:40 archlucas razer_mount[323]: Device_ID 0003:1532:0203.0003
Dec 15 18:50:40 archlucas razer_mount[324]: Modprobing razerkbd
Dec 15 18:50:40 archlucas systemd-udevd[234]: Process 'razer_mount razerkbd 0003:1532:0203.0003' failed with exit code 1.
`

~/.local/share/openrazer/logs/razer.log :
`Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | INFO     | Initialising Daemon (v2.1.1). Pid: 24760
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
[... 31 lines cut ...]
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaverChroma -------------------- (1532:0207)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2011 ------------------------- (1532:0013)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2013 ------------------------- (1532:0039)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
[... cut ...]
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
Dec 15 19:39:11 archlucas openrazer-daemon[24760]: 2017-12-15 19:39:11 | razer                          | INFO     | Serving DBus
`
Thank you.

z3ntu, 2017-12-15:
1. Never run the daemon with sudo.
2. Output of `sudo dkms install openrazer-driver/1.0.0; sudo dkms status` please

OverSurge, 2017-12-15:
1. Ok, wasn't sure.
2. Module openrazer-driver/1.0.0 already installed on kernel 4.14.5-1-ARCH/x86_64
openrazer-driver, 1.0.0, 4.14.5-1-ARCH, x86_64: installed

z3ntu, 2017-12-15:
Then `sudo modprobe razermouse` and `lsusb | grep 1532`

OverSurge, 2017-12-15:
Bus 003 Device 002: ID 1532:0203 Razer USA, Ltd BlackWidow Chroma
Bus 003 Device 005: ID 1532:0034 Razer USA, Ltd

OverSurge, 2017-12-15:
2nd one is a Taipan

z3ntu, 2017-12-15:
Have you tried to reboot already?

OverSurge, 2017-12-15:
Yep, several times, gonna do it one more time

z3ntu, 2017-12-15:
Okay then `sudo modprobe razerkbd; sudo modprobe razerfirefly; lsmod | grep razer`. It seems that razer_mount is failing with modprobing but I can't spot an issue in the info you sent...

OverSurge, 2017-12-15:
Found the problem : The USB port on the Blackwidow Chroma, as I unplugged, it worked. I think there is something that doesn't work at this point.

OverSurge, 2017-12-15:
razerfirefly           40960  0
razermouse             77824  0
razerkbd               69632  0
hid                   126976  6 razermouse,hid_generic,usbhid,razerkbd,razerfirefly,hid_plantronics
usbcore               282624  13 rndis_host,usbnet,razermouse,usbhid,razerkbd,snd_usb_audio,ehci_hcd,cdc_ether,razerfirefly,xhci_pci,snd_usbmidi_lib,xhci_hcd,ehci_pci
(I send it despite it works now)

tonylambiris, 2017-12-16:
FWIW I use a BlackWidow Chroma just fine w/ Wayland at work. Did you add yourself to the plugdev group?

z3ntu, 2017-12-16:
@oversurge can I close this now?

OverSurge, 2017-12-16:
Yes, added myself to the plugdev group, just to precise, the device plugged in the USB port is my phone on RNDIS aka USB Ethernet, don't know why it causes the driver to fail, but the issue is from here, i rebooted many times with an without it, and it worked only when it was unplugged, but you can close the issue, I have others working USB ports.

closed 2017-12-16 by z3ntu