openrazer/openrazer #476: BlackWidow TE Chroma v2 - No backlight until password entered.

etrigan63, 2017-12-16:
Running on Ubuntu 17.10 Gnome. Openrazer latest version and Polychromatic to manage. When I first installed the keyboard, it had a breathing mode that ran even when the screensaver kicked in. Now with Openrazer, I get all of the functions but the keyboard is dark until I enter my password. The log shows it suspending and waking up normally but the wakeup event should happen before the password is entered, not after. Kind of the point of an illuminated keyboard.

terrycain, 2017-12-16:
@etrigan63 Yeah That's an issue of whenever the screensaver daemon sends out dbus events, not much we can do

lah7, 2017-12-16:
There is a way to turn off this functionality altogether - edit this file: `~/.config/openrazer/razer.conf`

```
# Turn off the devices when the systems screensaver kicks in
devices_off_on_screensaver = False
```

etrigan63, 2017-12-16:
That will work just fine.

Sent from Mailspring (https://link.getmailspring.com/link/local-789eab8f-336c@Gojira/0?redirect=https%3A%2F%2Fgetmailspring.com%2F&recipient=reply%2B00701c40ee0c091b3180f22bc0f076b4ae1019faff193d9992cf00000001164d179792a169ce10d81f03%40reply.github.com), the best free email app for work

closed 2017-12-19 by z3ntu