openrazer/openrazer #477: Is it possible to layer effects on top of custom color schemes?

0tcmct0, 2017-12-17:
My question is as the title states. So far I've only either been able to get custom color schemes mapped, or have effects such as reactive, ripple, etc. I can't seem to figure out how to do both at the same time. Is there a way to do this? Thank you for your time.

0011001011, 2018-07-03:
Any update on this ?

z3ntu, 2018-07-06:
@0011001011 
> Thanks for this amazing software, that's so much fun to use. To have more fun, and following the previously closed issue (stale),  is it possible to activate some effects layers on top of a custom  colorscheme ? Ex: Reactive effect on some keys and custom colors on  others. https://github.com/openrazer/openrazer/issues/477

@0011001011 
> If there is no deep hardware driver-related issue for this I may be able to implement it myself if someone can provide me some insights

Copied from #615

z3ntu, 2018-07-06:
It should be possible to do this software-side (I mean in the daemon) but you would need to code the standard hardware effects (wave, breath, static, etc) in software. Then you can of course mix them together and create whatever you want.

0011001011, 2018-07-08:
Coding the standard effect in the daemon would be quite inefficient, probably slow to refresh smoothly, and not very battery efficient as it means a python calculation every 0.1s at least.
But the standards effect are already coded on a much lower level, closer to the hardware right ?

Actually this question seems stupid unless you see that the officiel Razer driver on Windows does that kind of layer thing. I will look into that.

terrycain, 2018-07-08:
@0011001011 Most effects were done in hardware. And correct, the Daemon could be adjusted to do them but it would be running a tight loop running every 0.05 seconds. The daemon does currently do a multi coloured ripple effect

0011001011, 2018-07-09:
Then the challenge here is to instruct the hardware to do some effect (wave, spectrum, and reactive, if not ripple) on a specific mapping of the matrix of the keys.
Dont have time this week but I will definitively check that

closed 2018-12-06 by stale[bot]

mind-overflow, 2020-10-16:
Hey, any update on this? Still seems to be an issue...

bluzukk, 2021-09-17:
> Hey, any update on this? Still seems to be an issue...

I don't think so. Have a look at this: https://gist.github.com/bluzukk/2f5ce1d21bcafbf6dd70d0b8f95a30f1