openrazer/openrazer #478: Repo URL for openSUSE incorrect

fschueller, 2017-12-18:
While manually adding the repository on openSUSE 42.3 (as described on https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta) it throws 

`Download (curl) error for 'https://download.opensuse.org/repositories/hardware:razer/openSUSE_Leap_42.3/hardware:razer.repo':
Error code:  Bad URL                                                                                                                                                                                                                         
Error message: Protocol http not supported or disabled in libcurl`

Directly adding the URL from [here](https://download.opensuse.org/repositories/hardware:/razer/openSUSE_Leap_42.3/hardware:razer.repo) reveals the previous URL seems to be wrong. 

Please verify and possibly correct it.

z3ntu, 2017-12-18:
Uhm it sounds like you have a broken libcurl installed but I can try it later in my VM.

fschueller, 2017-12-18:
I haven't checked that, it's true. The second URL works for me though so I thought it might be worth noting.

z3ntu, 2017-12-18:
I know that it's not this projects fault as we don't control the infrastructure of the opensuse build service :)

z3ntu, 2017-12-19:
I can reproduce the issue on openSUSE Leap 42.3... Please bother openSUSE so they fix whatever they have broken 😉

closed 2017-12-19 by z3ntu