openrazer/openrazer #479: Razer Blade Stealth logo_led_state inverted

joonatoona, 2017-12-19:
1 turns the logo off, and 0 turns it on.

z3ntu, 2017-12-19:
What version of the driver do you have? Also `lsusb | grep 1532` please

joonatoona, 2017-12-19:
2.1.1

`Bus 001 Device 004: ID 1532:0205 Razer USA, Ltd Blade Stealth`

z3ntu, 2017-12-19:
Do you know if it worked before release 2.1 / can you test if it worked before 2.1?

joonatoona, 2017-12-19:
I'm not sure how to downgrade the package, but I'm pretty sure it didn't work at all before 2.1. (Last time I tried the logo was off no matter what I did)

lezed1, 2017-12-20:
I tested this on my RBS Late 2016 (`1532:0220`). I unfortunately cannot test on the `0205`.

I can confirm the logo state is flipped as reported v2.1.1. It looks like the support for my laptop is [commented out](https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L1040) 😢. I'll open up a PR to fix that.

I tried v2.0.0 and there is no option for logo control at all in DBus, so it's hard to tell if this regressed.

@z3ntu: The original RBS that @joonatoona has is inverted in the driver. I can un-invert it to fix this issue if you would like.

z3ntu, 2017-12-20:
Well if that device doesn't have it inverted, yes, remove the "invert fix" for that device.

lezed1, 2017-12-20:
I added the un-invert for the `0205` Stealth to the PR.

samr28, 2018-01-25:
Can confirm that the logo LED is inverted on my Blade Stealth Mid 2017. Running the newest version of openrazer.

z3ntu, 2018-01-25:
@samr28 usb pid please (= output of lsusb)

samr28, 2018-01-25:
```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f3:2379 Elan Microelectronics Corp. 
Bus 001 Device 004: ID 1532:022d Razer USA, Ltd 
Bus 001 Device 003: ID 1bcf:2c9a Sunplus Innovation Technology Inc. 
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

z3ntu, 2018-01-27:
@samr28 Please test #480 (instructions are there)

z3ntu, 2018-03-30:
#480 is merged.

bkonsela, 2020-03-29:
Sorry to bump a dead issue, but this problem still exists on the Blade Stealth 2016.

`$ lsusb  | grep Razer
Bus 001 Device 004: ID 1532:0220 Razer USA, Ltd Blade Stealth (2016)
`

z3ntu, 2020-03-29:
@bkonsela Can you try removing this line https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L1370 from the driver, rebuild and load the updated kernel module and test again?

bkonsela, 2020-03-29:
That seems to have done the trick. The function seems to work properly now.

Can I go ahead and create a branch/submit a pull request? Or will you do it?

z3ntu, 2020-03-29:
@lezed1 originally reported that his model (also 1532:0220, in this issue further above) needs the led inverted so I'm not sure how to proceed here... I'm guessing that maybe a firmware update (that not everybody has of course) has fixed the issue or something?

bkonsela, 2020-03-29:
Strange. I know that the fix was applied to 1532:0205. I can check on my end for a firmware update.

lezed1, 2020-03-30:
I fever having this issue, but not much more than that anymore. I don't use this much since I moved away from Arch, so I don't have a quick way to test this either. If both our devices seem to be different, I have no particular reason to think mine is "correct".

bkonsela, 2020-03-30:
So how should we proceed?

z3ntu, 2020-03-30:
Can you both send the firmware version of your device `/sys/bus/hid/drivers/razerkbd/0*/firmware_version` if possible?

bkonsela, 2020-03-30:
```
$ cat /sys/bus/hid/drivers/razerkbd/0003:1532:0220.0003/firmware_version
v1.0
```

...That sounds old, doesn't it

0x6d6f7468, 2020-08-05:
@z3ntu I know this is rather old, but my system has had a recent update of openrazer and polychromatic, and the logo controls are still inverted. If you'd rather, I'm happy enough just to forget about it.