openrazer/openrazer #483: Razer Tartarus V2

dflament, 2017-12-21:
Hello,

i'm really new at this so please, if i sound like a noob, don't laugh :-) Asking kindly support for the pretty new Tartarus V2, only difference to the previous model is the addition of a scroll wheel and Chroma. Many thanks already for any effort made...

**lsusb -d 1532: -v**

> Bus 003 Device 008: ID 1532:022b Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x022b 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Tartarus V2
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

**ls -lah /dev/input/by-id/**

> lrwxrwxrwx 1 root root   9 Dec 21 21:39 usb-Razer_Razer_Tartarus_V2-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Dec 21 21:39 usb-Razer_Razer_Tartarus_V2-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Dec 21 21:39 usb-Razer_Razer_Tartarus_V2-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Dec 21 21:39 usb-Razer_Razer_Tartarus_V2-if02-mouse -> ../mouse0

**Model name, number & Product code**

> Model name & number : RZ07-0227, Gaming Keypad
Product code : RZ07-02270100

JPGs to follow tomorrow as i need my wife's windowz computer

dflament, 2017-12-22:
Screenshots from Synapse v3 Beta, hope this is good enough?

[image]
[image]
[image]
[image]
[image]

dflament, 2018-01-02:
do you also need a record packet captures of the USB traffic or is the above information more then enough?

WhitePatches, 2018-01-07:
Also volunteering to help get this device supported. Functionally almost identical to Orbweaver Chroma, with the addition of a scrollwheel in the place of button 20, and button 20 on the thumbrest. I'm hoping that it uses the same protocol :)

WhitePatches, 2018-01-07:
Digging around in the Synapse 3 files on Windows, there's a reference to "NostromoV2"

manisjp, 2018-03-08:
What do you need for the pcap?

z3ntu, 2018-03-08:
@manisjp basically the same as in the other issue. At least for the beginning :)

monsieuraa, 2018-04-11:
I buy a tartarus V2 to, how I can help ?

z3ntu, 2018-04-11:
@monsieuraa We need packet captures (see https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) of the functions that the device can do (like switching lighting effect, changing brightness)

dflament, 2018-05-13:
ok, after many months i finally got the time to mess around a bit with reverse engineering. Can someone please look at this and tell me if its ok? i did the brightness and the quick effects in a seperate pcap.

Please tell me if i did something wrong or not and if you need more...

[tartarusv2_brightness.pcapng.gz](https://github.com/openrazer/openrazer/files/1998752/tartarusv2_brightness.pcapng.gz)
[tartarusv2_quickeffects.pcapng.gz](https://github.com/openrazer/openrazer/files/1998754/tartarusv2_quickeffects.pcapng.gz)

z3ntu, 2018-05-13:
pcaps look good, thanks.

twiddletec, 2018-06-05:
Just wanted to note, that we now also own the V2. 
Drop me note if you need someone testing...

ly0, 2018-06-17:
Really need for this. Waiting for more detail.

nelyafinwe, 2018-08-01:
do you need help with this? i have this device as well (0x022b)

0mnipotent, 2018-08-20:
I've got a Tartarus V2 1532:022b, 
I can follow instructions and provide data, or i can test what is designed for this.
looking forward to anything :)

Altoholic, 2018-10-23:
I have a Tartarus Chroma v2 too, if i can help.

bboyles303, 2018-12-03:
Did this ever happen in a testing/beta release? I have a Tartarus Chroma V2 as well and would love to use it.

triggerhappytog, 2018-12-18:
Would like to buy the Tartarus V2 for programming purposes, has it been tested?

danielrobbins, 2019-01-24:
@WhitePatches  I've tried adding the Tartarus V2 and treating it as the equivalent of an OrbWeaver, and at least in polychromatic setting various color stuffs doesn't work. So it may use a different protocol. My commit is here in case anyone else would like to play with it: https://github.com/funtoo/openrazer

If I get more time I may play with USB packet captures to see what's needed to get it working.

SunnyBTC, 2019-03-13:
Hey. I've got the tartarus v2 also. Happy to help out? Was anyone able to work out how it works?

0mnipotent, 2019-03-14:
Hi there,

Nah I never heard anything back :/
Which is a shame.
Let me know if you have any luck!

trialism, 2019-04-17:
I have this device too. Can we expect support in the near future?

SunnyBTC, 2019-04-18:
I'm with you, praying!

dimascyriaco, 2019-04-20:
I have one v2, too. If you guys need any help just ask.

laaameche, 2019-04-25:
I have no coding abilities but like to help to have this device (that I own) working on linux.
Moving from Windows to Linux is not easy with that type of problem :/ but thank you for the other devices that the project support.

rporter220, 2019-04-30:
I also have a Tartarus V2 and would like to provide any assistance that is needed.

trialism, 2019-05-02:
This is the most popular keypad in my country and there's barely any other type in the stores. It'd be great if linux would support such a popular device.

Darkslide77, 2019-05-03:
I also have a Razer Tartarus V2 and am willing to help with any development.

lannet2000, 2019-05-30:
I also got one tartarus V2 , I'm a programmer if I can help drop me a message.
Thanks

freedrone, 2019-08-02:
I think the problem right now is to find some one who is familiar with the code base, and have time to deal with the development process

trialism, 2019-08-07:
I tried to make this work with https://github.com/funtoo/openrazer by merging 2.6.0, updating the `install_files/appstream/io.github.openrazer.openrazer.metainfo.xml` file and creating a new `pylib/openrazer/_fake_driver/razertartarusv2.cfg` file(based on razerorbweaver/tartarus). Here is the diff: https://pastebin.com/tyyyKfSw

Polychromatic threw this:

Device Not Recognised
A Razer product is connected to your computer, but the daemon didn't detect it.
It is possible your Razer product is not bound to the driver properly or it is not yet supported.

@z3ntu can you help me?

z3ntu, 2019-08-08:
@trialism Are you sure the dkms kernel module is built with the new code? And the udev rule installed and the daemon running from the patched source? From looking at the code, I don't see anything obviously wrong.

trialism, 2019-08-08:
@z3ntu I purged openrazer and then reinstalled it with this repo: https://github.com/trialism/openrazer-meta-aur and these commands:

```sh
sudo pacman -U openrazer-driver-dkms-2.6.0-1-any.pkg.tar.xz
sudo pacman -U openrazer-daemon-2.6.0-1-any.pkg.tar.xz
sudo pacman -U openrazer-meta-2.6.0-1-any.pkg.tar.xz
sudo pacman -U python-openrazer-2.6.0-1-any.pkg.tar.xz
```

It can detect the device now I just can't do anything with it. Neither the spectrum nor the breath nor the static effects nor the brightness settings work.

z3ntu, 2019-08-08:
@trialism You have to me the `case` statements you added in the driver to different locations then (or add case statements).

At least for brightness (`razer_attr_write_set_brightness`) I can tell you, that you need a case with `report = razer_chroma_extended_matrix_brightness(VARSTORE, ZERO_LED, brightness);` according to the pcaps. If that doesn't work you can try adding `report.transaction_id.id = 0x1F;` as well. That should hopefully work then.

trialism, 2019-08-09:
@z3ntu I didn't really understand the first part so I just added a few cases: https://github.com/trialism/openrazer/commit/2edb30c61e2406fc3a6623b0e251e47c67d8aa34 - it didn't work.

The brightness works sometimes(probably a GUI issue:) but the slider always shows the zero value.

lannet2000, 2019-08-09:
Thanks @trialism  and @z3ntu for youe effort, see you are making a Aur package , nice :)
Please let me/us know if and how we can help you.

z3ntu, 2019-08-09:
> The brightness works sometimes

What do you mean by that?

trialism, 2019-08-09:
I meant that it doesn't always react to the slider's changes and when it does the slider will go back to the starting position no matter what I do. Even if the slider is at 0 I can pull it from right to left to turn down the brightness(can't turn off) and if I want to increase I can pull from left to right but at both cases the slider will show zero brightness while the actual brightness will change.

z3ntu, 2019-08-09:
Try adding into `razer_attr_read_set_brightness` the code `report = razer_chroma_extended_matrix_get_brightness(VARSTORE, ZERO_LED);` (and the transaction_id thing if it's needed)

trialism, 2019-08-09:
Many thanks, the brightness works now: https://github.com/trialism/openrazer/commit/bb33887bb98c8cda33da8e1c771dd300bd199934

What do you think, can we solve the static-effects issue? Is that a special case usually or is the tartarus_v2 doing something differently?

@lannet2000 someone else already provided the relevant pcap files - I'm not familiar with those but if you want to, you could help us tweak the branch.

z3ntu, 2019-08-09:
Unfortunately the packet captures don't contain any useful information for the effects unfortunately because Synapse 3 just controls everything software-side but in OpenRazer we're using the "hardware" effects (which the firmware on the device can do), so it will be a case of trial and error I think.

trialism, 2019-08-10:
So good old brute-force. Would it be possible to change the color of specific keys?

trialism, 2019-08-10:
And is it the driver which decides which effects are available or do we get feedback from the firmware?

z3ntu, 2019-08-10:
> Would it be possible to change the color of specific keys?

That's achieved with the custom frame effect.

> And is it the driver which decides which effects are available or do we get feedback from the firmware?

Sometimes the device says "no I can't do that", sometimes it just quietly does nothing so no, we don't really get any usable responses from the firmware.

z3ntu, 2019-08-10:
One more thing I can tell you from the pcaps is, that you need to use `razer_chroma_extended_matrix_effect_custom_frame` with rows 0 to 3 and columns 0 to 5 for custom frame (the first thing you put into the driver, and the dimensions are put into the daemon, so it will be `MATRIX_DIMS = [4, 6]` I think)

trialism, 2019-08-10:
Thanks for the quick reply and the info!

trialism, 2019-08-10:
I added a few effects but only the "none" effect works. The frontends don't show the keys and the keypad. https://github.com/trialism/openrazer/commit/9f07174a92e8653842ccc6da28e1f6010b6afe93

z3ntu, 2019-08-10:
You need to tell the daemon that `'set_custom_effect', 'set_key_row',` are available, otherwise the frontends don't know anything about that. About the other effects, again it will have to be a game of brute force. Likely the `report.transaction_id.id = 0x1F;` statement will be needed for the effects as well.

trialism, 2019-08-10:
Good news, the following effects work: none, wave, reactive, breath(all types), static.

Gaming mode appeared in the gui but I don't know how should I use it.

I tried to add macro support but the keyboard is still not visible and I can't change the color of individual keys either.

trialism, 2019-08-10:
I will also try to make the spectrum, starlight and ripple effects work.

trialism, 2019-08-10:
Starlight and spectrum also work. I removed game-mode because it crashed the tray applet. The ripple effect is supported according to the "master guide" but it doesn't work(probably an issue with the matrix support?). And there is a fire effect too but there is no such helper function.

z3ntu, 2019-08-10:
Ripple and Fire effects are done in software.

For the custom frame matrix thing, I don't see anything obviously wrong in your changes. I'm not sure what logic polychromatic uses to display a button there or not, so you can try to compile RazerGenie (see https://github.com/z3ntu/RazerGenie#how-to-compile) and use `meson -Dinclude_matrix_discovery=true` instead of the meson command listed in the readme.

trialism, 2019-08-10:
I tried that but it didn't display the keypad or the keys.

trialism, 2019-08-10:
Is it possible that we need `razer_chroma_extended_ [g,s]et_led_state` functions instead of `razer_chroma_standard_[g,s]et_led_state` functions to make macros and the keypad display work? So far, all the lighting effects needed the `extended` helpers and didn't work with the `standard` ones.

tonyperiod, 2019-09-09:
i got the V2 as well, moved to linux for better modeling but without a working keypad can't do much... programmers pls do your magic, we love you

tonyperiod, 2019-09-09:
i have the V2 as well.. i have 0 experience with programming and i just installed linux. but if there is anything i can do to help tell me. i have subscribed to this discussion, i need the tartarus for 3d modeling

erlendbirketvedt, 2019-09-30:
Any updates on progress here? Anything we can do to help out?

jere-leppanen, 2019-10-01:
seconded, I switched to linux and can't use my tartarus v2 how I want now. It seems like it's the only razer device not supported by openrazer.

trialism, 2019-10-01:
I sold mine so I won't work on it anymore.  Make sure to grab the changes from my profile before I delete them. Most of the lighting options worked when I tried them.

jere-leppanen, 2019-10-14:
> I sold mine so I won't work on it anymore. Make sure to grab the changes from my profile before I delete them. Most of the lighting options worked when I tried them.

I can't figure out how to even get my tartarus v2 device recognized in polychromatic with you files. I download it, do make, or sudo make install etc. and always saying Device Not Recognised in polychromatic.

trialism, 2019-10-15:
@jere-leppanen it's an Arch/AUR package.

[10 more replies]

commit 8b459653 (2020-03-05): Add support for Razer Tartarus V2

Fixes #483
closed 2020-03-05 by z3ntu