openrazer/openrazer #484: Issues on installing openrazer on Ubuntu 16.04

apsusanto0, 2017-12-26:
Hey there. I have a problem when running `sudo apt install openrazer-meta` on Ubuntu 16.04. I think the problem lies with my kernel version (4.15), but I'm not sure.

Here's the output on the terminal

`apsusanto@apsusanto:~$ sudo apt install openrazer-meta
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  dkms openrazer-daemon openrazer-doc openrazer-kernel-modules-dkms
  python3-daemonize python3-evdev python3-numpy python3-openrazer
  python3-pyudev python3-setproctitle xautomation
Suggested packages:
  python-daemonize-doc python-numpy-doc python3-dev python3-nose
  python3-numpy-dbg
The following NEW packages will be installed:
  dkms openrazer-daemon openrazer-doc openrazer-kernel-modules-dkms
  openrazer-meta python3-daemonize python3-evdev python3-numpy
  python3-openrazer python3-pyudev python3-setproctitle xautomation
0 upgraded, 12 newly installed, 0 to remove and 89 not upgraded.
Need to get 0 B/2,063 kB of archives.
After this operation, 11.3 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package python3-daemonize.
(Reading database ... 338151 files and directories currently installed.)
Preparing to unpack .../python3-daemonize_2.4.7-xenial_all.deb ...
Unpacking python3-daemonize (2.4.7-xenial) ...
Selecting previously unselected package dkms.
Preparing to unpack .../dkms_2.2.0.3-2ubuntu11.5_all.deb ...
Unpacking dkms (2.2.0.3-2ubuntu11.5) ...
Selecting previously unselected package openrazer-kernel-modules-dkms.
Preparing to unpack .../openrazer-kernel-modules-dkms_2.1.1~ubuntu16.04.1_amd64.deb ...
Unpacking openrazer-kernel-modules-dkms (2.1.1~ubuntu16.04.1) ...
Selecting previously unselected package python3-pyudev.
Preparing to unpack .../python3-pyudev_0.16.1-3_all.deb ...
Unpacking python3-pyudev (0.16.1-3) ...
Selecting previously unselected package python3-evdev.
Preparing to unpack .../python3-evdev_0.4.7-0ubuntu4_amd64.deb ...
Unpacking python3-evdev (0.4.7-0ubuntu4) ...
Selecting previously unselected package python3-setproctitle.
Preparing to unpack .../python3-setproctitle_1.1.8-1build2_amd64.deb ...
Unpacking python3-setproctitle (1.1.8-1build2) ...
Selecting previously unselected package xautomation.
Preparing to unpack .../xautomation_1.09-1_amd64.deb ...
Unpacking xautomation (1.09-1) ...
Selecting previously unselected package openrazer-daemon.
Preparing to unpack .../openrazer-daemon_2.1.1~ubuntu16.04.1_amd64.deb ...
Unpacking openrazer-daemon (2.1.1~ubuntu16.04.1) ...
Selecting previously unselected package openrazer-doc.
Preparing to unpack .../openrazer-doc_2.1.1~ubuntu16.04.1_all.deb ...
Unpacking openrazer-doc (2.1.1~ubuntu16.04.1) ...
Selecting previously unselected package python3-numpy.
Preparing to unpack .../python3-numpy_1%3a1.11.0-1ubuntu1_amd64.deb ...
Unpacking python3-numpy (1:1.11.0-1ubuntu1) ...
Selecting previously unselected package python3-openrazer.
Preparing to unpack .../python3-openrazer_2.1.1~ubuntu16.04.1_all.deb ...
Unpacking python3-openrazer (2.1.1~ubuntu16.04.1) ...
Selecting previously unselected package openrazer-meta.
Preparing to unpack .../openrazer-meta_2.1.1~ubuntu16.04.1_all.deb ...
Unpacking openrazer-meta (2.1.1~ubuntu16.04.1) ...
Processing triggers for man-db (2.7.5-1) ...
Setting up python3-daemonize (2.4.7-xenial) ...
Setting up dkms (2.2.0.3-2ubuntu11.5) ...
Setting up openrazer-kernel-modules-dkms (2.1.1~ubuntu16.04.1) ...
Loading new openrazer-driver-1.0.0 DKMS files...
First Installation: checking all kernels...
Building only for 4.15.0-999-generic
Building initial module for 4.15.0-999-generic
ERROR (dkms apport): kernel package linux-headers-4.15.0-999-generic is not supported
Error! Bad return status for module build on kernel: 4.15.0-999-generic (x86_64)
Consult /var/lib/dkms/openrazer-driver/1.0.0/build/make.log for more information.
-e 
ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'
Setting up python3-pyudev (0.16.1-3) ...
Setting up python3-evdev (0.4.7-0ubuntu4) ...
Setting up python3-setproctitle (1.1.8-1build2) ...
Setting up xautomation (1.09-1) ...
Setting up openrazer-daemon (2.1.1~ubuntu16.04.1) ...
Setting up openrazer-doc (2.1.1~ubuntu16.04.1) ...
Setting up python3-numpy (1:1.11.0-1ubuntu1) ...
Setting up python3-openrazer (2.1.1~ubuntu16.04.1) ...
Setting up openrazer-meta (2.1.1~ubuntu16.04.1) ...

z3ntu, 2017-12-26:
I guess you manually installed Linux 4.15 as it's not available normally in any Ubuntu distribution. 
Can you send us the contents of the file `/var/lib/dkms/openrazer-driver/1.0.0/build/make.log` ?

apsusanto0, 2017-12-26:
Yes, I did install it manually because I heard the newest kernel could fix the caps lock issue Blade Stealth has with Ubuntu (which it didn't).

Here's my `make.log`

`make -C /lib/modules/4.15.0-999-generic/build SUBDIRS=/var/lib/dkms/openrazer-driver/1.0.0/build/driver modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-999-generic'
Makefile:932: *** "Cannot generate ORC metadata for CONFIG_UNWINDER_ORC=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel".  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-999-generic'
Makefile:25: recipe for target 'driver' failed
make: *** [driver] Error 2`

z3ntu, 2017-12-26:
Please read the message given in the make.log 

> "Cannot generate ORC metadata for CONFIG_UNWINDER_ORC=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"

and do that.
Btw you can make codeblocks like
```
this
```
by using triple backticks (```) at the start and the end of a block

z3ntu, 2018-03-30:
Closing as there is no response.

closed 2018-03-30 by z3ntu