openrazer/openrazer #486: Failed to synchronize cache for repo 'hardware_razer', disabling. on Fedora 27

chrisdk22, 2017-12-28:
after running  dnf install openrazer-meta kernel-devel I get this error
Failed to synchronize cache for repo 'hardware_razer', disabling.
no match for argument: openrazer-meta

I can't figure out what to do to get it to download.

z3ntu, 2017-12-28:
Can you try what was suggested here https://ask.fedoraproject.org/en/question/92573/failed-to-synchronize-cache-for-repo/?answer=92629#post-id-92629 ?

chrisdk22, 2017-12-28:
Thanks for the suggestion. I tried it but still got no luck.

chrisdk22, 2017-12-28:
Failed to synchronize cache for repo 'hardware_razer', disabling.              ] ---  B/s |   0  B     --:-- ETA
Last metadata expiration check: 1:32:12 ago on Thu 28 Dec 2017 09:46:18 AM MST.
No match for argument: openrazer-meta
Package kernel-devel-4.14.8-300.fc27.x86_64 is already installed, skipping.
Error: Unable to find a match

chrisdk22, 2017-12-29:
when i run dnf check-update I get this:
Fedora 27 - x86_64 - Updates                                                                                            6.0 MB/s |  14 MB     00:02    
Fedora 27 - x86_64                                                                                                      6.0 MB/s |  58 MB     00:09    
Failed to synchronize cache for repo 'hardware_razer', disabling.                                                     ] ---  B/s |   0  B     --:-- ETA

z3ntu, 2017-12-29:
I can't reproduce this on a fresh Fedora 27 install.

chrisdk22, 2017-12-29:
I'm sorry so i only should have to run those two commands correct?
 dnf config-manager --add-repo http://download.opensuse.org/repositories/hardware:razer/Fedora_27/hardware:razer.repo
 dnf install openrazer-meta kernel-devel

I really appreciate your help

z3ntu, 2017-12-30:
Yes, that are the commands I have used.
This seems to be a local issue so please use your google-fu to find the solution yourself as I can do nothing more than googling myself and giving you things to try.
I guess there are forums for Fedora where you can ask your question and hopefully get the answer.
Good luck with that!

closed 2017-12-30 by z3ntu

chrisdk22, 2017-12-31:
I got it working. I had to take it home and run it, something weird with the internet firewalls or something at the office.
Thanks again for all the help and suggestions