openrazer/openrazer #487: [Arch/Manjaro] Ornata Chroma not recognized (in Polychromatic)

allienworks, 2017-12-29:
### My setup

- latest (Arch-based) Manjaro with Gnome
- Ornata Chroma keyboard (supported by openrazer)
- openrazer & polychromatic

### My installation

1. I have installed openrazer via AUR by installing `openrazer-daemon` `openrazer-drivers-dkms` `openrazer-meta` and `python-openrazer`
2. for the GUI `polychromatic`
3. added user to plugdev via `sudo gpasswd -a $USER plugdev`

### The problem

Upon startup, Polychromatic Controller gives me error "Device not recognized":

[image]

### Logs, outputs

```
$ lsusb | grep '1532:'
Bus 002 Device 004: ID 1532:021e Razer USA, Ltd
```
```
$ ls -lah /sys/bus/hid/drivers/
total 0
drwxr-xr-x 3 root root 0 28. pro 20.15 ./
drwxr-xr-x 4 root root 0 28. pro 20.15 ../
drwxr-xr-x 2 root root 0 28. pro 20.15 hid-generic/
```
```
$ ls -lah /sys/bus/hid/drivers/hid-generic/
total 0
drwxr-xr-x 2 root root    0 28. pro 20.15 ./
drwxr-xr-x 3 root root    0 28. pro 20.15 ../
lrwxrwxrwx 1 root root    0 29. pro 01.30 0003:1532:021E.0003 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.8/2-1.8:1.0/0003:1532:021E.0003/
lrwxrwxrwx 1 root root    0 29. pro 01.30 0003:1532:021E.0004 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.8/2-1.8:1.1/0003:1532:021E.0004/
lrwxrwxrwx 1 root root    0 29. pro 01.30 0003:1532:021E.0005 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.8/2-1.8:1.2/0003:1532:021E.0005/
lrwxrwxrwx 1 root root    0 29. pro 01.30 0003:E0FF:0005.0001 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.0/0003:E0FF:0005.0001/
lrwxrwxrwx 1 root root    0 29. pro 01.30 0003:E0FF:0005.0002 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.1/0003:E0FF:0005.0002/
--w------- 1 root root 4,0K 29. pro 01.30 bind
lrwxrwxrwx 1 root root    0 29. pro 01.30 module -> ../../../../module/hid_generic/
--w------- 1 root root 4,0K 29. pro 01.30 new_id
--w------- 1 root root 4,0K 28. pro 20.15 uevent
--w------- 1 root root 4,0K 29. pro 01.30 unbind
```

- `razer.log`: https://pastebin.com/jQA9DPq0
- `openrazer-daemon -Fv`: https://pastebin.com/Zyypu0nL

I've read all the Troubleshooting info here, also browsed all the issues here and on Polychromatic – can't find anything helpful. Can anyone point me to the right direction?

I'm posting this here as this seem to be an issue with the driver.

z3ntu, 2017-12-29:
Please run `sudo dkms install openrazer-driver/1.0.0; sudo dkms status`

allienworks, 2017-12-29:
```
$ sudo dkms install openrazer-driver/1.0.0; sudo dkms status
[sudo] password for allien: 

Creating symlink /var/lib/dkms/openrazer-driver/1.0.0/source ->
                 /usr/src/openrazer-driver-1.0.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.9.72-1-MANJARO cannot be found at
/usr/lib/modules/4.9.72-1-MANJARO/build or /usr/lib/modules/4.9.72-1-MANJARO/source.
openrazer-driver, 1.0.0: added
```

So this is the kernel header issue, right?

z3ntu, 2017-12-29:
Yes you have to install the kernel headers for your kernel. On Arch it's `linux-headers` but on Manjaro the kernel packages are named slightly different. I think I saw something like kernel package: linux49 and then kernel headers: linux49-headers or something.

z3ntu, 2017-12-29:
Also see https://wiki.archlinux.org/index.php/Dynamic_Kernel_Module_Support

allienworks, 2017-12-29:
@z3ntu great, thanks for help! I've installed the `linux49-headers` and it all works after restart. Thanks very much!