openrazer/openrazer #488: Request for support: Razer Imperator (classic)

hawajka76, 2018-01-01:
Hello, 
I noticed that only the Razer Imperator 2012 seems to be supported. Do you think there would be a chance of implementing support for the older Razer Imperator model?
I'm appending further info below:

**lsusb -d 1532: -v**

us 001 Device 008: ID 1532:0017 Razer USA, Ltd RZ01-0035 Laser Gaming Mouse [Imperator]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0017 RZ01-0035 Laser Gaming Mouse [Imperator]
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     101
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1

**ls -lah /dev/input/by-id/**

total 0
drwxr-xr-x 2 root root 120 Jan  1 12:49 .
drwxr-xr-x 4 root root 460 Jan  1 13:02 ..
lrwxrwxrwx 1 root root   9 Dec 31 18:24 usb-J5PF77DKZ_Lenovo_EasyCamera-event-if00 -> ../event9
lrwxrwxrwx 1 root root  10 Jan  1 12:49 usb-Razer_Razer_Imperator-event-mouse -> ../event13
lrwxrwxrwx 1 root root  10 Jan  1 12:49 usb-Razer_Razer_Imperator-if01-event-kbd -> ../event14
lrwxrwxrwx 1 root root   9 Jan  1 12:49 usb-Razer_Razer_Imperator-mouse -> ../mouse1

**Screenshots** - this model is not supported by Razer synapse so the screenshots are from the older program which is used to configure the mouse

[image]
[image]
[image]
[image]
[image]

**Model no.**
RZ01-0035

**Product no.**
RZ01-00350100

z3ntu, 2018-01-01:
Could you please record packet captures of the USB traffic with Wireshark (and save the file as `.pcapng`, not `.pcap` and one file/capture per bullet point) of:
* Change DPI from 1000 to 5600 and back
* Change poll rate from 125Hz to 500Hz to 1000Hz
* Change scroll wheel on->off->on
* Change logo on->off->on

If you are not familiar with capturing USB traffic, take a look at our [wiki page](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) and follow it until the middle of phase 3 (you can ignore everything after the screenshot of Wireshark in Phase 3).

Thanks!

hawajka76, 2018-01-02:
Thanks for offering to help! I'm attaching the requested data. I've added ".txt" to the name of each file to be able to upload them. Please let me know in case anything else would be helpful.

[125 to 500 to 1000 poll rate.pcapng.txt](https://github.com/openrazer/openrazer/files/1598862/125.to.500.to.1000.poll.rate.pcapng.txt)
[1000 to 5600dpi.pcapng.txt](https://github.com/openrazer/openrazer/files/1598863/1000.to.5600dpi.pcapng.txt)
[logo_on_off_on.pcapng.txt](https://github.com/openrazer/openrazer/files/1598864/logo_on_off_on.pcapng.txt)
[scroll_wheel_on_off_on.pcapng.txt](https://github.com/openrazer/openrazer/files/1598865/scroll_wheel_on_off_on.pcapng.txt)

mirh, 2020-06-15:
Up. 
Also, who should I nag to get "dpi buttons actually being read as microphone on/off" sorted out?