openrazer/openrazer #489: Razer Ornata Chroma blinking/flashing M

Bednar87, 2018-01-01:
The Macro indicator located above the numeric keypad keeps flasing in red and I cannot seem to be able to stop it. FN + F9 doesn't have any effect.

Any ideas?

Daemon log:

`2018-01-01 23:32:05 | razer                          | INFO     | Initialising Daemon (v2.2.0). Pid: 4747
2018-01-01 23:32:05 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerMambaTE ---------------------------- (1532:0046)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerNagaChroma ------------------------- (1532:0053)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaverChroma -------------------- (1532:0207)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2011 ------------------------- (1532:0013)
2018-01-01 23:32:05 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2013 ------------------------- (1532:0039)
[... cut ...]
Jan 01 23:14:38 arch-Bednar razer_mount[595]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:021E.0004/
Jan 01 23:14:38 arch-Bednar razer_mount[597]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:021E.0004/
Jan 01 23:14:38 arch-Bednar razer_mount[609]: Finished binding 0003:1532:021E.0006
Jan 01 23:14:38 arch-Bednar razer_mount[610]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:021E.0006/
Jan 01 23:14:38 arch-Bednar razer_mount[612]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:021E.0006/
`

Bednar87, 2018-01-01:
Possibly related to https://github.com/polychromatic/polychromatic/issues/123

z3ntu, 2018-03-30:
Is this still an issue in the latest version?

Ryoga85, 2018-07-01:
It does happen with my Cynosa Chroma on the current version as well and it makes sense because both Ornata Chroma and Cynosa Chroma don't have any M keys (so you need to be able to bind Macros to any regular key in order to use them). Just pressing Fn+F9 again stops recording a Macro but it doesn't stop the M-LED from blinking.

closed 2018-11-28 by stale[bot]

dimyme, 2019-03-31:
if self._current_macro_bind_key is None:
-                        # Restrict macro bind keys to M1-M5
-                        if key_name not in ('M1', 'M2', 'M3', 'M4', 'M5'):
-                            self._logger.warning("Macros are only for M1-M5 for now.")
-                            self._recording_macro = False
-                            self._parent.setMacroMode(False)
-                        else:
                            self._current_macro_bind_key = key_name
                            self._parent.setMacroEffect(0x00)

will make stuff work on Ornata keyboard model which has no "M..." extra keys.

Ryoga85, 2021-08-22:
Hi, I had the same problem with my Cynosa Chroma and because neither any of the comments here nor on the linked post could help, I decided to look deeper into this.

Turned out when you start the macro mode by pressing Fn+F9, the M-LED is set to flashing mode with "self._parent.setMacroEffect(0x01)" in line 411 of file /usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py

However, when you press Fn+F9 again, the macro mode is ended, but someone forgot to turn the LED off again.
So I added "self._parent.setMacroEffect(0x00)" between line 423 and 424, which fixed the issue for me.

The whole thing starting from line 401 now looks like this:

                # Macro FN+F9 logic
                if key_name == 'MACROMODE':
                    self._logger.info("Got macro combo")

                    if not self._recording_macro:
                        # Starting to record macro
                        self._recording_macro = True
                        self._current_macro_bind_key = None
                        self._current_macro_combo = []

                        self._parent.setMacroEffect(0x01)
                        self._parent.setMacroMode(True)

                    else:
                        self._logger.debug("Finished recording macro")
                        # Finish recording macro
                        if self._current_macro_bind_key is not None:
                            if len(self._current_macro_combo) > 0:
                                self.add_kb_macro()
                            else:
                                # Clear macro
                                self.dbus_delete_macro(self._current_macro_bind_key)
                        self._recording_macro = False
                        self._parent.setMacroEffect(0x00)
                        self._parent.setMacroMode(False)

z3ntu, 2021-08-24:
Please submit a PR if you have a solution, and also avoid necro-bumping old issues