openrazer/openrazer #490: Options missing on Razer Abyssus 2014

zbagz, 2018-01-02:
Hey there, I've upgraded from Debian Jessie to Debian Stretch 9.3 x64 and now my mouse is moving awfully slow. and the same mouse worked just fine on debian jessie without installing anything as far as I can remember. 

I've tried a couple of different things, including using a different usb port, xinput set-prop, Xorg -configure with a couple different of settings I found on google in an attempt to increase the acceleration but nothing seems to work. So, I thought this could be something related to drivers and installed openrazer.

The problem is that the only option that shows up on my panel is the brightness button to turn off/on the logo light. (I've tried with Polychromatic and RazerGenie but the same thing happens, here's a screenshot https://i.imgur.com/mMfrQup.png). 

My mouse is a Razer Abyssus 2014 | 1532:0042, and according to the docs it *should* work, but I'm not sure, maybe this is a known issue or just something expected due to being a rather old mouse.

Thanks.

z3ntu, 2018-01-02:
Do you have access to a Windows Virtual Machine? It seems that DPI was never implemented for this mouse (see e.g. #204). If yes, it would be great if you could make screenshots of the options in Synapse (you can ignore the logo led toggle and key remapping (if available)).

z3ntu, 2018-01-23:
Yes, if you could make packet captures when changing the dpi slider and the polling rate setting, that would make it very easy (see https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol)

tweelix, 2020-04-15:
Hopefully reviving this as This is my favourite mouse.
I tried capturing packets however just changing the dpi slider doesn't seem to generate any packets at all. 
Therefore the first file dpi-1500-100.pcapng is me moving the mouse at 1500dpi, changing the dpi to 100 and then moving the mouse again. Packet No 0-94 where at 1500dpi and packets after that at 100.

I did manage to capture the packets for changing the polling rate setting. The second file polling_rate_500-1000-125 changes the polling rate first to 500, then to 1000 and finally to 125.

The files can be found here:
[captures.zip](https://github.com/openrazer/openrazer/files/4481885/captures.zip)

I'm kind of new to this so if there's anything else I can do to help let me know.

z3ntu, 2020-04-19:
@tweelix Thanks. I also only see polling rate in the captures so it looks like Synapse does DPI in software on Windows (and not in hardware).

You can try adding `CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_poll_rate);` to https://github.com/openrazer/openrazer/blob/master/driver/razermouse_driver.c#L3138 and `device_remove_file(&hdev->dev, &dev_attr_poll_rate);` to https://github.com/openrazer/openrazer/blob/master/driver/razermouse_driver.c#L3552 and `'get_poll_rate', 'set_poll_rate',` to https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/mouse.py#L330 , then poll rate should be available in the front ends.
If you need any help with this, just ask. :)

tweelix, 2020-04-19:
@z3ntu So that works for the polling rate. I can submit a PR if you'd like.

However the DPI mystery deepens. Whenever I finish my testing in the virtualbox windows VM the sensitivity of my mouse is really off, specifically it's way too high. It stays this way until I connect it back to my windows desktop PC after which the sensitivity is back to how it normally is. So maybe it is storing something on the mouse? Mind you it might just be a bug with X11 of some sort.
 
I actually have two Abyssus mice so I tested one that has been connected to the windows VM and one has come straight from being plugged into my windows pc. Both times set the DPI was set to 1500. The one that's been connected to the VM has much higher sensitivity. All their xinput-props are the same however. Wonder if you have any idea what may be going on here.

tweelix, 2020-04-19:
> However the DPI mystery deepens. Whenever I finish my testing in the virtualbox windows VM the sensitivity of my mouse is really off, specifically it's way too high. It stays this way until I [...]

Never mind, if I just unplug the mouse and plug it back in again it fixes the sensitivity. So it seems to just be either a X11 or XFWM issue. I'll just submit a PR for the Polling Rate

z3ntu, 2020-04-24:
#1105 is merged. Closing.

closed 2020-04-24 by z3ntu