openrazer/openrazer #491: Forward / Back button only working while holding right click on 1532:0045 Mamba Chroma (wireless)

xervox, 2018-01-02:
Hi everyone,

I am running into an issue in Mint 18.3 with a recently bought Razer Mamba "ID 1532:0045" in wireless mode.

Forward and Back buttons of the mouse are only reacting when I hold the right button:
verified with "xev | grep button".
_state 0x0, button 3, same_screen YES
    state 0x400, button 8, same_screen YES  <-- back
    state 0x400, button 9, same_screen YES  <-- forward_

when I hit only the button without right click I get this message on both:

_KeymapNotify event, serial 40, synthetic NO, window 0x0,
    keys:  4294967261 0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0_

xorg.conf looks like this:

_Section "InputDevice"
    Identifier     "Mouse0"
    Driver         "mouse"
    Option         "Protocol" "auto"
    Option         "Device" "/dev/psaux"
    Option         "Emulate3Buttons" "no"
    Option         "ZAxisMapping" "4 5"
EndSection_

I am running:
`openrazer-daemon  2.2.0-1~319~ubuntu16.04.1  amd64 OpenRazer Device Daemon`
`openrazer-kernel-modules-dkms 2.2.0-1~319~ubuntu16.04.1 amd64  OpenRazer Peripheral Drivers`
`openrazer-meta   2.2.0-1~319~ubuntu16.04.1 all Metapackage for OpenRazer components`

`cat /sys/bus/hid/drivers/razermouse/0003\:1532\:0045.000C/firmware_version`
`v1.0`

Any ideas how to fix this?

thx a lot

z3ntu, 2018-01-02:
Did you try uninstalling the driver and rebooting to see if it fixes the issue?

xervox, 2018-01-03:
I got a temporary workaround by running the following command:

_xmodmap -e "pointer = 1 2 3 4 5 6 7 9 8 10 11 12 13"_

_xev | grep button_
is still behaving like before with right click only, but at least my browser is accepting forward and back now.

so I guess the basic issue remains. 
I am playing a little bit with _imwheel_ now and reading up on mouse mappings. Still strange so my previous mouse was not behaving like that.

closed 2018-06-02 by stale[bot]