openrazer/openrazer #494: Support Void Linux

osum4est, 2018-01-05:
Void Linux is a great minimal distro like Arch, but uses runit instead of systemd. An install package would be great!

z3ntu, 2018-01-07:
I can't get Void Linux installed in my VM (based on virt-manager). I've tried the images `void-live-i686-20171007-lxqt.iso` and `void-live-x86_64-20171007-lxqt.iso` (i686 variant in BIOS mode and the x86_64 variant in both BIOS and UEFI mode) but the installer always hangs while downloading packages... If you can tell me how I can make it not hang, I can take a look at it again.

osum4est, 2018-01-07:
Interesting. I just successfully installed `void-live-x86_64-20171007-lxqt.iso` with BIOS in QEMU w/ libvert on an Arch Linux host. I used the command `qemu-system-x86_64 -cdrom void-live-x86_64-20171007-lxqt.iso -boot order=d -drive file=disk,format=raw` where disk was a 4G raw disk.

Did you choose Local or Network for the installation source? If you choose local it shouldn't try to download anything during the installation (i think). You could try going into Network during the installation to see if you can connect from there. I'm not sure what host OS you are using, but on Arch I needed to install and enable ebtables and dnsmasq before networking would work correctly.

If the problem is not network connectivity, but it just getting stuck for some reason then I'm not too sure. Maybe more ram or disk space? 

If you still can't figure it out, I'm uploading my QEMU disk right now, and i'll post the link once it finishes.

Hopefully that helps!

z3ntu, 2018-01-07:
Yes, I used network sources during installation. And I am also using Arch for the host system. And it was definitely not a network problem because some packages download successfully and then everything hangs so I can't even move the mouse cursor inside the VM (or switch VTY etc)

osum4est, 2018-01-07:
Huh. How much ram and disk did you give it? I gave mine 4G of disk and 1G of ram. Maybe something in here would help? https://wiki.archlinux.org/index.php/QEMU#Virtual_machine_runs_too_slowly

z3ntu, 2018-01-07:
10GB disk, 2GB ram and 2 CPU...

z3ntu, 2018-01-07:
And with local sources it still hangs 
[image]

osum4est, 2018-01-07:
The only thing I can think of is to try the base `void-live-x86_64-20171007.iso` image without any gui. 

If possible you could try VirtualBox/VMWare. There's also an irc, #voidlinux on freenode that may be able to help. Not sure how much you want to put into this though.

z3ntu, 2018-01-07:
To be honest, if I have such troubles installing a Linux distro, it doesn't really feel like I want to spend my time with it...

osum4est, 2018-01-07:
Fair enough. Went smoothly for me, but everyone's system is different... I'll attempt to build it myself (and probably fail!)

z3ntu, 2018-01-07:
Packaging seems to be similar to Arch Linux so you might be able to take the PKGBUILD from the AUR and change a few things (such as method names).

osum4est, 2018-01-08:
Never packaged anything before, but I'll certainly give it a shot.

z3ntu, 2018-01-08:
Now I've tried again and the installation surprisingly worked but then in the installed system running `sudo xbps-source -Syu` (might contain a typo) hanged it again just like my previous attempts. I've recoded a screencast of it (I actually 'hoped' it would hang at installation time again ^^): https://youtu.be/3uZ_0ilpB80

osum4est, 2018-01-08:
I just tried a full upgrade on my VM and it completed successfully. I'm even running plasma like you. It could be your lack of swap but I doubt it since 2GB of ram should be plenty.

Is the GUI frozen as well, or just the xbps process? Does it still happen after rebooting and trying again?

Also, I'm no void linux expert - only just started using it about a week ago, so I honestly don't know why this is happening. I'm happy to try to package it though, don't have anything else to do today.

z3ntu, 2018-01-08:
Actually upgrading via the command line worked.. I can give you the `template` file when I have something remotely useful running, then you can gather all the dependencies that are needed (and package them if needed, idk how void is with python dependencies)

z3ntu, 2018-01-08:
You can base off of https://github.com/z3ntu/void-packages/commit/cc386c43fcc3da497c75db7efcc0eda052f9dcc3 . Somehow it seems the pkg_install function of the subpackages doesn't get called so the files never get moved. For 'getting started' instructions you can visit https://wiki.voidlinux.eu/Xbps-src . Also it might be beneficial to you to join our Telegram / Matrix chat, for Telegram go to https://t.me/joinchat/A1E4RT9TxhnxyetdQCSj7Q and for Matrix go to https://matrix.to/#/#openrazer:matrix.org

osum4est, 2018-01-08:
Well that was fast! I'll take a look at it and see what I can do!

z3ntu, 2018-01-28:
PR is at https://github.com/voidlinux/void-packages/pull/10653

z3ntu, 2018-01-28:
The PR was just merged.

closed 2018-01-28 by z3ntu