openrazer/openrazer #495: Support for Razer Cynosa Pro Bundle

artokarin, 2018-01-06:
OpenRazer 2.2 didn't see anything.
I found little tool for keyboard configuration https://github.com/john-sevsk/cynpro , but it can't configurate mouse.
```
$ sudo lsusb -d 1532: -v

Bus 003 Device 002: ID 1532:020d Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x020d 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Cynosa Pro Bundle
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

Bus 003 Device 007: ID 1532:004d Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x004d 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
```

$ ls -lah /dev/input/by-id/ | grep -i razer
lrwxrwxrwx 1 root root   9 янв  6 12:42 usb-Razer_Razer_Cynosa_Pro_Bundle-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 янв  6 12:42 usb-Razer_Razer_Cynosa_Pro_Bundle-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 янв  6 12:42 usb-Razer_Razer_Cynosa_Pro_Bundle-if02-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 янв  6 12:42 usb-Razer_Razer_Cynosa_Pro_Bundle-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 янв  6 12:42 usb-Razer_Razer_DeathAdder-event-mouse -> ../event16
lrwxrwxrwx 1 root root   9 янв  6 12:42 usb-Razer_Razer_DeathAdder-mouse -> ../mouse2

```

Sdlion, 2018-05-18:
Currently trying to set up a VirtualBox Windows 10 machine to capture the usb communication (I can't get VBox to passthrough the keyboard and mouse, they appear as generic devices and Synapse doesn't recognize them).
But for the moment here is someone that already has some samples of the communication protocol for the three preset colors and the cycling breathing mode.
https://github.com/john-sevsk/cynpro

It's to note that this keyboard seems to only have 2 backlight colors. The red byte is ignored when is modified. With a rgb(255,0,0) no light is lit on the keyboard.

Sdlion, 2018-05-21:
Here are the wireshark captures of the lighting and gamer different modes of the Cynosa Pro keyboard.
I'll capture later the macro programming sequence and DeathAdder mouse lighting and programming options
[Cynosa-Pro-Lighting-Gaming.zip](https://github.com/openrazer/openrazer/files/2024342/Cynosa-Pro-Lighting-Gaming.zip)
.

Sdlion, 2018-05-22:
Here are the captures of the macro programming sequences for Cynosa Pro Keyboard.
With and without recorded delays; deleting "test" and "test2" profiles; after recording adding (before and after) delays (0.080, 0.050); after recording adding keystroke with delay before (1.20).
[Cynosa-Pro-Macros.zip](https://github.com/openrazer/openrazer/files/2027533/Cynosa-Pro-Macros.zip)

DeathAdder2000 mouse wireshark captures
Lighting: breathing, static three fixed colors, brightness control; for each scroll wheel and mouse logo.
Performance: Can be set to 1, 2, 3, 4 or 5 sensitivity stages (only recorded setting 3 and 5 values); change of polling rate (from a list of 125, 500, 1000 values) and change to an arbitrary DPI.
[DeathAdder2000-Lighting.zip](https://github.com/openrazer/openrazer/files/2027551/DeathAdder2000-Lighting.zip)
[DeathAdder2000-Performance.zip](https://github.com/openrazer/openrazer/files/2027552/DeathAdder2000-Performance.zip)

Screen captures:

**cynosapro-customizeui**
[image]

**cynosapro-gamingmodeui**
[image]

**cynosapro-lightingui**
[image]

**cynosapro-macros-addingafterrecordingoptions**
[image]

**cynosapro-macrosemptyui**
[image]

**deathadder2000-customizebuttonfunctions**
[image]

**deathadder2000-customizeui**
[image]

**deathadder2000-lightingnoneselected**
![deathadder2000-lightingnoneselected](https://user-images.githubusercontent.com/7925567
/40376355-bb422f96-5db3-11e8-8438-e4b402e5a54f.png)

**deathadder2000-lightinguistaticandbreathingoptions**
[image]

**deathadder2000-performanceui**
[image]

**deathadder2000-sensitivitystagesui**
[image]

z3ntu, 2018-07-11:
I don't think so.

artokarin, 2019-02-08:
Any progress here?

FuzzyGamers, 2019-03-21:
Any progress? Or any work around to force the OS to use regular Cynosa build?

MahmoodSaroo, 2019-04-15:
Any progress here?

FuzzyGamers, 2019-04-15:
> Any progress here?

Someone opened a new thread on this the developer needs a PCAP for it all. I'd love to do it but after googling it still have no idea how to.

z3ntu, 2019-04-16:
It is described on this wiki page: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

FuzzyGamers, 2019-04-21:
Crap I dont have a spare keyboard to give this a go. :( This seems unfortunately complicated as well. I'll see about getting a spare and trying to make this happen.

atasro2, 2021-12-26:
the mouse from this bundle seems to be the same as #1188 with the exception of the product number which is RZ01-01450100 as opposed to RZ01-01450200, this is further solidified with the fact that by changing the device IDs associated with #1188 to from 0x4F to 0x4B the mouse is able to communicate the openrazer and display information from itself and the lighting can be adjusted through the driver

z3ntu, 2022-01-15:
@atasro2 If that works, can you submit a PR to add this device?

PseudoDistant, 2022-03-24:
Sorry, I should've checked the tags, anyone get the keyboard working yet?

artokarin, 2022-03-30:
@PseudoDistant 3.2.0-1~701 still doesn't support it. You can use https://github.com/john-sevsk/cynpro

artokarin, 2022-03-30:
My mouse died 2 years ago. I think kbd will die before it will receive support in openrazer :smiley:

PseudoDistant, 2022-03-31:
;-;