openrazer/openrazer #496: Can't control speed for wave effect in Razer Firefly

da-edra, 2018-01-07:
I can control the brightness and direction of the effect but not the speed of the effect.

The speed is too fast and does not allow to fully appreciate the effect, is there a way to control the speed?

I'm using the latest version from the AUR.

z3ntu, 2018-01-07:
Which frontend are you using? I think this should be an issue in the repository of that frontend.

da-edra, 2018-01-07:
polychromatic, but the same happens with razergenie and razercommander.

lezed1, 2018-01-07:
@da-edra can you control the speed in Windows? I don't remember that option personally, but I may be wrong.

da-edra, 2018-01-07:
I don't use Windows but yes, you can do that there.

z3ntu, 2018-01-07:
Yes, the daemon doesn't support a speed argument for setWave. If you can find us a screenshot (Google or similar) from Synapse where it's possible we can talk about it. But as wave is a hardware effect, synapse also has to have the effect.

da-edra, 2018-01-07:
I see... I think I confussed it with Corsair Polaris where you can adjust the speed.
Thank you.

closed 2018-01-07 by da-edra

zkarcher, 2019-03-21:
@lezed1 On the Razer Blade 15 2018, the speed can be controlled in Windows. (Screenshot attached.) Is there any chance this issue can be re-opened?

[image]

lezed1, 2019-03-21:
@zkarcher That looks like the custom effect gui which calculates all the color values in software and then sends the colors many times pet second to the keyboard. This is different from telling the keyboard once to play a standard wave effect, and not needing to send each frame of the animation. The hardware effect is not as configurable.

The daemon's setWave controls the hardware effect. If you want more control, you can calculate colors in software and use the daemon to send each frame several times a second.

The daemon itself doesn't support these complex effects, but all hope is not lost! I have some examples in this repo regarding how to use the advanced Python API. I also have 2 separate libraries designed to easily run complex effects over time. If you are interested in writing your own effects, I'm happy to help since that is my personal favorite use of OpenRazer.

zkarcher, 2019-03-21:
@lezed1 Thanks for explaining this!

I just wrote code using the Python API, and I was able to produce the exact effect I wanted in software. Gist: https://gist.github.com/zkarcher/b27493baf3a026d6b9dce9069eb4ceec

I'd like to check out your libraries for running complex effects, do you mind sharing the repos?

lezed1, 2019-03-25:
The better of the two is this project: https://github.com/lezed1/color-full. I don't have any info about how to use it, so feel free to ask me questions. It's split into Effects (complex effects like you want) and Devices (backends to display the effects - a software visualizer is in progress). It's still in a hacky stage, but I have been able to run everything. I'm also happy to merge in things if you modify it. Hopefully it's useful for you to use!