openrazer/openrazer #497: Razer Mamba 2012 not detected

Steve-Arch, 2018-01-07:
Hello,

I have a Razer Mamba 2012 (Wired) that I can't see to get working.
It appears to be a supported device -

[stephen@Archey ~]$ lsusb
Bus 003 Device 002: ID 1532:0024 Razer USA, Ltd Mamba

My user is in the correct groups -
[stephen@Archey ~]$ groups stephen
wheel plugdev stephen

The Daemon also seems to load correctly without any errors being logged (See Daemon log file attached) However all I get in Polychromatic is "A Razer product is connected to your computer but the Daemon couldn't detect it"
[image]

[razer.log](https://github.com/openrazer/openrazer/files/1609716/razer.log)

My system info is here -

[stephen@Archey ~]$ uname -a
Linux Archey 4.14.11-1-ARCH #1 SMP PREEMPT Wed Jan 3 07:02:42 UTC 2018 x86_64 GNU/Linux

Thank you :)

z3ntu, 2018-01-07:
Did you install the kernel headers for your kernel? See https://wiki.archlinux.org/index.php/Dynamic_Kernel_Module_Support#Installation

Steve-Arch, 2018-01-07:
Oh wow, I am an idiot. 

I have DKMS installed but I forgot my headers!

Thank you for that z3ntu! 

Great driver by the way!

closed 2018-01-07 by Steve-Arch

z3ntu, 2018-01-07:
No problem, you're not the first person to have had this issue 😉