openrazer/openrazer #498: [Arch/Manjaro] Black Widow Chroma V2 not recognized

MrHappy45, 2018-01-08:
Hi there, I'm having problems with openrazer not willing to cooperate at all. I'm pretty newbie at Linux and everything, but I have a basic understanding of things. I installed openrazer through pacaur then installed polychromatic using yaourt. Then the app said that it didn't recognize any devices. So I tried starting the daemon using the console (`openrazer-daemon -Fv`) then it said that the user wasn't in the plugdev group, so I moved my user in it, but still no luck.

### My Setup

- Latest Majaro with Gnome
- Black Widow Chroma V2 which is supported by openrazer
- openrazer and polychromatic (I've tested other front-end apps aswell, same result)

### Logs & Outputs
```
$ lsusb | grep '1532:'
Bus 003 Device 003: ID 1532:0221 Razer USA, Ltd 
```
```
$ ls -lah /sys/bus/hid/drivers/hid-generic/
total 0
drwxr-xr-x 2 root root    0 Jan  7 21:22 .
drwxr-xr-x 3 root root    0 Jan  7 21:22 ..
lrwxrwxrwx 1 root root    0 Jan  7 22:28 0003:1038:1710.0005 -> ../../../../devices/pci0000:00/0000:00:1c.4/0000:04:00.0/usb5/5-2/5-2:1.0/0003:1038:1710.0005
lrwxrwxrwx 1 root root    0 Jan  7 22:28 0003:1038:1710.0006 -> ../../../../devices/pci0000:00/0000:00:1c.4/0000:04:00.0/usb5/5-2/5-2:1.1/0003:1038:1710.0006
lrwxrwxrwx 1 root root    0 Jan  7 22:28 0003:1038:1710.0007 -> ../../../../devices/pci0000:00/0000:00:1c.4/0000:04:00.0/usb5/5-2/5-2:1.2/0003:1038:1710.0007
lrwxrwxrwx 1 root root    0 Jan  7 22:28 0003:1532:0221.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.0/0003:1532:0221.0002
lrwxrwxrwx 1 root root    0 Jan  7 22:28 0003:1532:0221.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.1/0003:1532:0221.0003
lrwxrwxrwx 1 root root    0 Jan  7 22:28 0003:1532:0221.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.2/0003:1532:0221.0004
lrwxrwxrwx 1 root root    0 Jan  7 22:28 0003:17A0:0305.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9:1.3/0003:17A0:0305.0001
--w------- 1 root root 4.0K Jan  7 22:28 bind
lrwxrwxrwx 1 root root    0 Jan  7 22:28 module -> ../../../../module/hid_generic
--w------- 1 root root 4.0K Jan  7 22:28 new_id
--w------- 1 root root 4.0K Jan  7 21:22 uevent
--w------- 1 root root 4.0K Jan  7 22:28 unbind
```
```
$ openrazer-daemon -Fv
Starting daemon.
2018-01-07 22:30:49 | razer                          | INFO     | Initialising Daemon (v2.2.0). Pid: 13393
2018-01-07 22:30:49 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
[... 16 lines cut ...]
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2011 ------------------------- (1532:0013)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2013 ------------------------- (1532:0039)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2018-01-07 22:30:49 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
[... cut ...]
2018-01-07 22:30:49 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2018-01-07 22:30:49 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2018-01-07 22:30:49 | razer                          | INFO     | Serving DBus
```
### What I've tried
I've tried to follow this issue #487 and installed the linux49 kernel but still no luck
and that's about it, I've tried all day to fix it.

Thanks in advance for any help.
Cheers!

z3ntu, 2018-01-08:
What's the output of `sudo dkms install openrazer-driver/1.0.0` ?

MrHappy45, 2018-01-08:
`sudo dkms install openrazer-driver/1.0.0` outputs this:
```
Error! echo
Your kernel headers for kernel 4.14.12-1-MANJARO cannot be found at
/usr/lib/modules/4.14.12-1-MANJARO/build or /usr/lib/modules/4.14.12-1-MANJARO/source.

```

z3ntu, 2018-01-08:
There you go. Just follow what's written there and install your kernel headers.

closed 2018-01-08 by z3ntu