openrazer/openrazer #499: Magic version issue Fedora 27

andyfore, 2018-01-08:
After installing the various packages on my fresh Fedora 27 install, I found that the modules will not load.

Here is the error in the output from dmesg:
`[ 1378.469529] razercore: version magic '4.14.11-300.fc27.x86_64+debug SMP mod_unload ' should be '4.14.11-300.fc27.x86_64 SMP mod_unload '
[ 1384.330135] razerkbd: version magic '4.14.11-300.fc27.x86_64+debug SMP mod_unload ' should be '4.14.11-300.fc27.x86_64 SMP mod_unload '`

Here is the full uname output:
`Linux it205512.web.web.com 4.14.11-300.fc27.x86_64 #1 SMP Wed Jan 3 13:52:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

Here are the relevant kernel packages that are installed:
`[afore@it205512 ~]$ rpm -qa | grep kernel
kernel-4.14.11-300.fc27.x86_64
kernel-debug-devel-4.14.11-300.fc27.x86_64
kernel-core-4.14.11-300.fc27.x86_64
kernel-modules-4.13.9-300.fc27.x86_64
kernel-headers-4.14.11-300.fc27.x86_64
kernel-core-4.13.9-300.fc27.x86_64
kernel-modules-extra-4.14.11-300.fc27.x86_64
kernel-4.13.9-300.fc27.x86_64
libreport-plugin-kerneloops-2.9.3-1.fc27.x86_64
openrazer-kernel-modules-dkms-2.2.0-1.1.noarch
kernel-modules-4.14.11-300.fc27.x86_64
kernel-modules-extra-4.13.9-300.fc27.x86_64
abrt-addon-kerneloops-2.10.5-1.fc27.x86_64`

I gather the issue is the I am running a kernel with debugging symbols.  Solutions?

z3ntu, 2018-01-08:
Uninstall kernel-debug-devel, then run `sudo dkms remove openrazer-driver/1.0.0 --all`, `sudo dkms install openrazer-driver/1.0.0` and reboot

z3ntu, 2018-01-08:
Oh yeah and you're missing `kernel-devel` so install that.

z3ntu, 2018-01-28:
Closing as there was no answer.

closed 2018-01-28 by z3ntu