openrazer/openrazer #500: Unable to control Tartarus

scoony, 2018-01-10:
Hi guys,

I've got a Tartarus and i've "tried" to install openrazer on my Ubuntu 17.10 but not really working properly.

RazerGenie changed the light to pulsate but now i can't revert to static and no way to assignate anything to it.

Using command line in /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.2/0003:1532:0201.0003, i get an error if I try to cat matrix_effect_static error of input/output.

scoony, 2018-01-10:
"ps aux | grep razer" showed a process openrazer-daemon, i've killed it and if I try to "openrazer-daemon -Fv" i get a "Unable to create the pidfile"

scoony, 2018-01-10:
Gonna provide some infos

`
/sys/bus/hid/drivers
├── hid-generic
│   ├── 0003:045E:0745.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:045E:0745.0004
│   ├── 0003:045E:0745.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.1/0003:045E:0745.0005
│   ├── 0003:045E:0745.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.2/0003:045E:0745.0006
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
└── razerkbd
    ├── 0003:1532:0201.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:1532:0201.0001
    ├── 0003:1532:0201.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.1/0003:1532:0201.0002
    ├── 0003:1532:0201.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.2/0003:1532:0201.0003
    ├── bind
    ├── module -> ../../../../module/razerkbd
    ├── new_id
    ├── uevent
    └── unbind
`

scoony, 2018-01-10:
Daemon info:

● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-01-10 06:38:38 CET; 5h 9min ago
     Docs: man:openrazer-daemon(8)
 Main PID: 31417 (openrazer-daemo)
   CGroup: /user.slice/user-1000.slice/user@1000.service/openrazer-daemon.service
           └─31417 openrazer-daemon

janv. 10 09:55:08 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 09:55:08 | razer.device0                  | DEBUG    | DBus call get_device_type
janv. 10 09:55:08 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 09:55:08 | razer.device0                  | DEBUG    | DBus call get_device_name
janv. 10 09:55:08 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 09:55:08 | razer.device0                  | DEBUG    | DBus call get_device_name
janv. 10 09:55:08 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 09:55:08 | razer.device0                  | DEBUG    | DBus call get_brightness
janv. 10 09:55:08 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 09:55:08 | razer.device0                  | DEBUG    | DBus call get_brightness
janv. 10 09:55:08 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 09:55:08 | razer.device0                  | DEBUG    | DBus call keypad_profile_led_red
janv. 10 09:55:08 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 09:55:08 | razer.device0                  | DEBUG    | DBus call keypad_get_profile_led_green
janv. 10 09:55:08 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 09:55:08 | razer.device0                  | DEBUG    | DBus call keypad_get_profile_led_blue
janv. 10 09:55:08 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 09:55:08 | razer.device0                  | DEBUG    | DBus call get_firmware
janv. 10 10:03:34 Serveur-NAS openrazer-daemon[31417]: 2018-01-10 10:03:34 | razer.device0.keymanager       | DEBUG    | Macro String: DOWN

scoony, 2018-02-24:
No reply? ... up

z3ntu, 2018-02-24:
Anything in dmesg?

scoony, 2018-02-24:
What do you want me to search for?

z3ntu, 2018-02-24:
Something with razerkbd

scoony, 2018-02-24:
Here is what I've got:

```
scoony@Serveur-NAS:~$ dmesg | grep razer
[    5.676137] razerkbd 0003:1532:0201.0005: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Tartarus] on usb-0000:00:1d.0-1.6/input0
[    5.736071] razerkbd 0003:1532:0201.0006: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Tartarus] on usb-0000:00:1d.0-1.6/input1
[    5.738243] razerkbd 0003:1532:0201.0007: input,hidraw5: USB HID v1.11 Mouse [Razer Razer Tartarus] on usb-0000:00:1d.0-1.6/input2
scoony@Serveur-NAS:~$
```

z3ntu, 2018-02-24:
According to what I see in the outputs there should be nothing wrong...

scoony, 2018-02-25:
Is there any command lines I could try to test the deamon?

Enabling/Disabling the pulsate effect or anything?

closed 2018-08-27 by stale[bot]

coberry87, 2020-03-15:
I am having this exact issue. What do you need from me as my details (minus the bus location of course) line us with scoony