openrazer/openrazer #501: Ripple effect doesn't work on keyboard plugged in after starting the deamon

GiantBlargg, 2018-01-11:
When I plug in my keyboard after starting the deamon, ripple doesn't work, the lights on my keyboard are all off. Restarting the deamon with `systemctl --user restart openrazer-daemon` with the keyboard plugged in fixes the problem, but if I unplug and replug, it is a problem again. Reactive and all other effects work fine.

lah7, 2018-01-19:
It's due to the fact ripple is a **software effect** and the daemon must be running in order for it to play. Sadly, it's not currently persistent between sessions or hardware changes, i.e. if the keyboard is disconnected or the daemon starts again, then the effect won't start playing again until you trigger that effect again.

If this happens when you login and the keyboard is your only Razer device, you could run this command for the **random ripple**:

    python3 -c "import openrazer.client;a=openrazer.client.DeviceManager();a.devices[0].fx.ripple_random()"

z3ntu, 2018-01-30:
Fixed in #516

PR #516 (2018-04-01): Work around race condition when hotplugging
The problem is that udev throws 'add' events in a random order but the
current code needs the interface with our sysfs entries first, and then
the other interfaces which get added as 'additional interfaces'. Without
this, in most cases the interfaces get added in the wrong order,
resulting in ripple, game mode, macro functionality etc not working

Now the to-be-added devices get batched up and sorted and only then
really added.

A downside of this commit is that adding new devices is getting delayed by at least 2 seconds (maybe this time can be lower [... cut]
closed 2018-04-01 by z3ntu