openrazer/openrazer #503: Installed packages but not working

ivan7707, 2018-01-12:
Hello, I installed the package as per the instructions:

https://openrazer.github.io/#download 

in the Ubuntu section

I found files here: /var/lib/dkms/openrazer-driver

But when I run openrazer-meta in terminal, noting happens.  

Any ideas would be appreciated

z3ntu, 2018-01-12:
What's the output of `systemctl --user status openrazer-daemon.service` and of `python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` ?

ivan7707, 2018-01-12:
Hi z3ntu, 

systemctl --user status openrazer-daemon.service

Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)

python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"

ImportError: No module named 'openrazer'

ivan7707, 2018-01-12:
In case this helps, this is what I get when I run **locate openrazer**

/etc/apt/sources.list.d/openrazer-ubuntu-stable-xenial.list
/etc/apt/trusted.gpg.d/openrazer_ubuntu_stable.gpg
/etc/apt/trusted.gpg.d/openrazer_ubuntu_stable.gpg~
/etc/xdg/autostart/openrazer-daemon.desktop
/home/ivan/.config/openrazer
/home/ivan/.config/openrazer/razer.conf
/home/ivan/.local/share/openrazer
/home/ivan/.local/share/openrazer/logs
/home/ivan/.local/share/openrazer/logs/razer.log
/usr/bin/openrazer-daemon
/usr/lib/python3/dist-packages/openrazer
/usr/lib/python3/dist-packages/openrazer_daemon
/usr/lib/python3/dist-packages/openrazer/__init__.py
/usr/lib/python3/dist-packages/openrazer/_fake_driver
/usr/lib/python3/dist-packages/openrazer/client
/usr/lib/python3/dist-packages/openrazer/_fake_driver/__init__.py
/usr/lib/python3/dist-packages/openrazer/client/__init__.py
/usr/lib/python3/dist-packages/openrazer/client/constants.py
/usr/lib/python3/dist-packages/openrazer/client/debug.py
/usr/lib/python3/dist-packages/openrazer/client/device.py
/usr/lib/python3/dist-packages/openrazer/client/devices
/usr/lib/python3/dist-packages/openrazer/client/fx.py
/usr/lib/python3/dist-packages/openrazer/client/macro.py
/usr/lib/python3/dist-packages/openrazer/client/devices/__init__.py
/usr/lib/python3/dist-packages/openrazer/client/devices/keyboard.py
/usr/lib/python3/dist-packages/openrazer/client/devices/mice.py
/usr/lib/python3/dist-packages/openrazer/client/devices/mousemat.py
/usr/lib/python3/dist-packages/openrazer_daemon/__init__.py
/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services
/usr/lib/python3/dist-packages/openrazer_daemon/device.py
/usr/lib/python3/dist-packages/openrazer_daemon/hardware
/usr/lib/python3/dist-packages/openrazer_daemon/keyboard.py
/usr/lib/python3/dist-packages/openrazer_daemon/misc
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/__init__.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/service.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/__init__.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/all.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/bw2013.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/deathadder_chroma.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/keypad.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/kraken.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/macro.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/mamba.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/mug.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/nagahex.py
/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/nagahexv2.py
/usr/lib/python3/dist-packages/openrazer_daemon/hardware/__init__.py
/usr/lib/python3/dist-packages/openrazer_daemon/hardware/core.py
/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py
/usr/lib/python3/dist-packages/openrazer_daemon/hardware/headsets.py
/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py
/usr/lib/python3/dist-packages/openrazer_daemon/hardware/mouse.py
/usr/lib/python3/dist-packages/openrazer_daemon/hardware/mouse_mat.py
/usr/lib/python3/dist-packages/openrazer_daemon/hardware/mug.py
/usr/lib/python3/dist-packages/openrazer_daemon/misc/__init__.py
/usr/lib/python3/dist-packages/openrazer_daemon/misc/battery_notifier.py
/usr/lib/python3/dist-packages/openrazer_daemon/misc/effect_sync.py
/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py
/usr/lib/python3/dist-packages/openrazer_daemon/misc/macro.py
/usr/lib/python3/dist-packages/openrazer_daemon/misc/ripple_effect.py
/usr/lib/python3/dist-packages/openrazer_daemon/misc/screensaver_monitor.py
/usr/share/openrazer
/usr/share/doc/openrazer-daemon
/usr/share/doc/openrazer-doc
/usr/share/doc/openrazer-kernel-modules-dkms
[... 66 lines cut ...]
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razerfirefly_driver.c
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razerfirefly_driver.h
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razerkbd_driver.c
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razerkbd_driver.h
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razerkraken_driver.c
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razerkraken_driver.h
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razermouse_driver.c
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razermouse_driver.h
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razermug_driver.c
/var/lib/dkms/openrazer-driver/1.0.0/build/driver/razermug_driver.h
/var/lib/dpkg/info/openrazer-daemon.conffiles
/var/lib/dpkg/info/openrazer-daemon.list
/var/lib/dpkg/info/openrazer-daemon.md5sums
/var/lib/dpkg/info/openrazer-daemon.prerm
/var/lib/dpkg/info/openrazer-doc.list
/var/lib/dpkg/info/openrazer-doc.md5sums
/var/lib/dpkg/info/openrazer-kernel-modules-dkms.list
/var/lib/dpkg/info/openrazer-kernel-modules-dkms.md5sums
/var/lib/dpkg/info/openrazer-kernel-modules-dkms.postinst
/var/lib/dpkg/info/openrazer-kernel-modules-dkms.prerm
/var/lib/dpkg/info/openrazer-meta.list
/var/lib/dpkg/info/openrazer-meta.md5sums
/var/lib/dpkg/info/python3-openrazer.list
/var/lib/dpkg/info/python3-openrazer.md5sums
/var/lib/dpkg/info/python3-openrazer.prerm

z3ntu, 2018-01-12:
Did you do anything with your python3 executable? I can't think of another reason why python doesn't the library which is installed according to the locate command. What's the output of `python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())"` ?

ivan7707, 2018-01-12:
Thank you Luca for helping me with this.  The result is this:

/home/ivan/anaconda3/lib/python3.5/site-packages

z3ntu, 2018-01-12:
Please reinstall python3 and remove any python installation that might be in your PATH variable.

ivan7707, 2018-01-13:
Hi Luca,  Reinstalled everything.  

I could not find any solution to my current issue in the issues.  

**systemctl --user status openrazer-daemon.service**

● openrazer-daemon.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)

**python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())"**

/usr/lib/python3/dist-packages

$ **python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"**

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer/client/__init__.py", line 22, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/openrazer/client/__init__.py", line 24, in __init__
    raise DaemonNotFound("Could not connect to daemon")
openrazer.client.DaemonNotFound: Could not connect to daemon

z3ntu, 2018-01-13:
Yes you have fixed your python installation. For Ubuntu 16.04 the command `systemctl --user status openrazer-daemon.service` is not used btw so that's expected to fail.

Can you run in your terminal `openrazer-daemon -Fv` and check if the daemon is exiting because of some reason? If it says `Serving DBus` at the end, everything is fine for the daemon.

ivan7707, 2018-01-13:
Thank you.  It worked.  The last line of openrazer-daemon -Fv showed serving DBus.

For anyone else who might land here, when I installed Polychromatic, it initially showed Device Not Recognized.  I ran this command to be sure it showed up, and it did (https://github.com/openrazer/openrazer#device-support I checked it against this list)

lsusb | grep '1532:'

The problem ended up being, just having to connect the keyboard to a different USB port, and it Polychromatic recognized the device.

I very much appreciate you taking your time to help me Luca.  I do not see a donations page, so let me just say Thank you and the openrazer team for being/supporting open source and making Ubuntu and Razer better!

closed 2018-01-13 by ivan7707

z3ntu, 2018-01-13:
Thanks :)
If you really want you can donate @terrycain (to motivate him to invest more time 😉 ) : https://github.com/openrazer/openrazer#contributions or me: https://www.paypal.me/z3ntu

ivan7707, 2018-01-13:
Right on. I'll kick you two a few buck for a beer/coffee on me.  Thanks again!

thelebdev, 2019-11-19:
Hey guys. I'm facing a similar issue. I've successfully installed the required packages but I can't find the APK in my apps list.

`systemctl --user status openrazer-daemon.service` returns
```
openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; ven
   Active: inactive (dead)
     Docs: man:openrazer-daemon(8)
lines 1-4/4 (END)
```

Any idea?

z3ntu, 2019-11-19:
@thelebdev Openrazer itself doesn't provide any GUI, you will have to install a so-called "frontend", e.g. polychromatic or RazerGenie.

thelebdev, 2019-11-20:
I had it installed on my old ubuntu machine as per the instructions on OpenRazer's website and it was working like a charm. I don't remember installing a standalone frontend for it?

z3ntu, 2019-11-20:
You definitely did install something other than openrazer if you had a GUI for configuring your Razer devices.

thelebdev, 2019-11-20:
I could have possibly forgotten that part then.
As for this issue, it's solved. I used RazerGenie. Thank you @z3ntu