openrazer/openrazer #504: Fedora 27 Fails to Synchronize with Repo

wcunning, 2018-01-14:
I installed the repository as instructed on the website. When I attempted to install the openrazer-meta package, I got the following dnf error on a fresh install of Fedora 27:

Failed to synchronize cache for repo 'hardware_razer', disabling.     --:-- ETA
Last metadata expiration check: 0:05:19 ago on Sun 14 Jan 2018 11:32:16 AM EST.

I then ping'd download.opensuse.org, and then opened the /etc/yum.repos.d/hardware\:razer.repo  file and opened the actual repository page in a browser. All of that indicates that everything is up online and functional, but dnf seems to not want to pull the repository for some reason. Any idea what's going on?

z3ntu, 2018-01-14:
[image]
It works for me in a Fedora 27 virtual machine. I am not too overly familiar with Fedora/dnf so maybe you can ask in the Fedora support forum (if that exists) what the reason could be.

wcunning, 2018-01-14:
I've asked it on their end, and we'll see what response I get. I will update when I have a resolution or if I get pointed back to download.opensuse.org.

https://ask.fedoraproject.org/en/question/116323/dnf-repo-failed-to-sync-cache/

jonesie3419, 2018-01-19:
I am having the same issue and am unable to find a solution either. I have done the same as above and confirmed that everything seems to be in working order as far as the repo link being a working link. I have also run: `dns clean all` followed by `dns update` and still get the same failed to synchronize error.

`
[jonesie@valhalla ~]$ dnf clean all
84 files removed
[jonesie@valhalla ~]$ dnf update
Docker CE Stable - x86_64                                                                                                                                                                                                                                                                                                                                    22 kB/s | 3.4 kB     00:00    
negativo17 - Nvidia                                                                                                                                                                                                                                                                                                                                         310 kB/s | 294 kB     00:00    
Fedora 27 - x86_64 - Updates                                                                                                                                                                                                                                                                                                                                7.1 MB/s |  16 MB     00:02    
Fedora 27 - x86_64                                                                                                                                                                                                                                                                                                                                          8.6 MB/s |  58 MB     00:06    
google-chrome                                                                                                                                                                                                                                                                                                                                                38 kB/s | 3.7 kB     00:00    
google-musicmanager                                                                                                                                                                                                                                                                                                                                          14 kB/s | 1.5 kB     00:00    
Node.js Packages for Fedora Linux 25 - x86_64                                                                                                                                                                                                                                                                                                               2.2 MB/s | 457 kB     00:00    
RPM Fusion for Fedora 27 - Free - Updates                                                                                                                                                                                                                                                                                                                   2.8 MB/s | 253 kB     00:00    
RPM Fusion for Fedora 27 - Free                                                                                                                                                                                                                                                                                                                             4.7 MB/s | 717 kB     00:00    
RPM Fusion for Fedora 27 - Nonfree - Updates                                                                                                                                                                                                                                                                                                                344 kB/s |  18 kB     00:00    
RPM Fusion for Fedora 27 - Nonfree                                                                                                                                                                                                                                                                                                                          1.8 MB/s | 163 kB     00:00    
slack                                                                                                                                                                                                                                                                                                                                                        33 kB/s |  30 kB     00:00    
Fedora 27 - x86_64 - VirtualBox                                                                                                                                                                                                                                                                                                                              88 kB/s |  34 kB     00:00    
Failed to synchronize cache for repo 'hardware_razer', disabling.
Last metadata expiration check: 0:00:00 ago on Fri 19 Jan 2018 03:03:19 PM EST.
Dependencies resolved.
Nothing to do.
Complete!
`

wcunning, 2018-01-20:
I have not gotten any traction other than trying dnf clean over on ask.fedoraproject.org. The fact that this is happening to someone else sort of inclines me to thinking that it's potentially a location issue and that bad mirrors are messing with us... Is there a good way to check that, maybe manually edit the mirror list that dnf is using? The verbose output I get says that all mirrors have been tried unsuccessfully.

z3ntu, 2018-01-21:
Does `wget http://download.opensuse.org/repositories/hardware:/razer/Fedora_27/repodata/71e652ba8c7f6d344c48c876ed85fe52fe47d0bb15e3b4480c45d491865f0364-primary.xml.gz` work for you?

fleekix, 2018-01-24:
Same issue here Fedora 27.

jonesie3419, 2018-01-24:
@z3ntu Sorry for the delay in response. Yes the wget works, I followed it up with the dnf update so you could see the two commands back to back, and I still get the weird synchronize cache error.

[image]

z3ntu, 2018-01-24:
@jonesie3419 Can you add `-v` to the dnf command? It should output something with `Cannot download` according to the dnf source code.

jonesie3419, 2018-01-24:
@z3ntu, sure here you go:

> Cannot download 'http://download.opensuse.org/repositories/hardware:/razer/Fedora_27/': Yum repo downloading error: Downloading error(s): repodata/71e652ba8c7f6d344c48c876ed85fe52fe47d0bb15e3b4480c45d491865f0364-primary.xml.gz - Cannot download, all mirrors were already tried without success; repodata/65ddfab8ea49c9a36fdf8bbf20a92ba2da760bef3bcd55719cb4cae1a0c574a5-filelists.xml.gz - Cannot download, all mirrors were already tried without success.

z3ntu, 2018-01-24:
Found that in a rather old bug report - can you try that?

> It seems like the connection of mirrors is closed because of timeout. Can you try to set `timeout=100` (or more) [1] into dnf.conf, please?
> 
> [1] http://dnf.readthedocs.org/en/latest/conf_ref.html#options-for-both-main-and-repo

jonesie3419, 2018-01-24:
Updated in both the repo and the /etc/dnf/dnf.conf file and still seeing the same error:

[image]

[image]

> Cannot download 'http://download.opensuse.org/repositories/hardware:/razer/Fedora_27/': Yum repo downloading error: Downloading error(s): [...]
repo: using cache for: nodesource

caffienetoolbar, 2018-01-28:
I just did a brand new install of Fedora 27 and am having the same error.

z3ntu, 2018-01-28:
I triggered a rebuild on OBS. Maybe that will fix something.

hyph3n, 2018-01-28:
Are any of you encountering this problem using a VPN?

I ran into a similar situation a few weeks ago and determined that the repo was killing my connection with an RST packet when attempting a dnf update--even though I could view the files/folders in a web browser no problem, and a dnf install worked just fine on a different system. 

I can't remember if I changed to a different VPN address or just temporarily turned it off on the failing machine, but I was eventually able to update normally. Haven't reproduced the issue since. ¯\\\_(ツ)\_/¯

z3ntu, 2018-01-28:
@hyph3n I activated a openvpn connection to my server on the host but in the guest VM dnf upgrade still works fine.

hyph3n, 2018-01-28:
In my case, the problem seemed to be related to the repo not liking one of my VPN provider's (PIA) addresses. As I said, switching to a different address cleared things up for me right away. YMMV :)

caffienetoolbar, 2018-01-28:
I'm not using a VPN.  I just literally installed Fedora 27 and tried to install.  Just a thought.  Maybe its failing the GPG check.  I ended up finding another page of yours with the binaries and think it installed it ok.  I tried to connect to the same repo for RazerGenie and that fails as well.

z3ntu, 2018-01-28:
I just installed a fresh VM with Fedora 27, ran the two commands listed on the website and it works without any problems. Sorry, but I really can't reproduce this.
Maybe some mirror on OBS makes problem. What's the country you're accessing the repo from? I'm from Austria.

jonesie3419, 2018-01-28:
I am not using a VPN either, I am in the United States. I will try a few other things to see if I can get it to work, I also have not tried since you did the rebuild of OBS.

hyph3n, 2018-01-28:
I just did the following:

1) `sudo dnf clean all`

2) `sudo dnf check-update`

`Failed to synchronize cache for repo 'hardware_razer', disabling.                                                           ] ---  B/s |   0  B     --:-- ETA`

3) Connect to VPN using **North American** IP address.

4) `sudo dnf clean all`

5) `sudo dnf check-update`

`Failed to synchronize cache for repo 'hardware_razer', disabling.                                                           ] ---  B/s |   0  B     --:-- ETA`

6) Connect to VPN using **German** IP address.

7) `sudo dnf clean all`

8) `sudo dnf check-update`

`hardware:razer (Fedora_26)                                                                                                     14 kB/s | 9.2 kB     00:00    `

I don't know why the repo would be rejecting certain IP addresses, but that certainly seems to be what's going on here. Sorry if I didn't explain myself better in my earlier comments.

z3ntu, 2018-01-28:
@hyph3n Thanks for testing. Could you write an email to the OBS mailing list (https://lists.opensuse.org/archive/opensuse-buildservice/) and explain the issue with basically the last comment you wrote? You can put me in CC (email is on my gh profile).

z3ntu, 2018-01-28:
Or send an email to admin at opensuse.org which opens a support ticket

wcunning, 2018-02-04:
It would appear that someone has done the communication with OBS, since I just noticed the repository not complaining when I updated packages today. I have successfully installed openrazer on Fedora 27 in Michigan, United States, North America. I'll leave it for the other people on this issue to chime in with their success or failure before closing the issue, but it looks resolved to me.

z3ntu, 2018-02-04:
@wcunning Yes, @hyph3n sent an email but I don't have access to the ticket made from that email but great, that it works now!

closed 2018-02-04 by z3ntu

usrix, 2018-03-16:
Was this officially resolved? Any instructions on how to make it work? Do you have the link of the ticket? I just tried and is not working... Thanks!

hyph3n, 2018-03-16:
Confirmed that the OBS repo is working correctly when I use a German exit node on my VPN, not working when I use a north american exit node. 

I am wondering if it would make sense to use COPR for Fedora (https://copr.fedorainfracloud.org/) since the OBS repo seems unreliable when accessed outside of Europe. I don't know exactly what would be involved or how big a lift it would be, but it might be worth investigation.

usrix, 2018-03-16:
I change my exit node to a German one, and it's working. Before I was in an north american exit node.