openrazer/openrazer #505: Original Generation Razer Mamba not supported

michaelschlies, 2018-01-17:
USB ID 1532:000A (wired)
USB ID 1532:000f (wireless)

z3ntu, 2018-01-17:
* Please post the output of
`lsusb -d 1532:000A -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

michaelschlies, 2018-01-18:
Model: ft703437pp
part: rz03-00120100-0000

lsusb
```

Bus 002 Device 024: ID 1532:000a Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x000a 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Mamba
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     101
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
/dev/input/by-id
```
total 0
drwxr-xr-x 2 root root 260 Jan 17 20:25 .
drwxr-xr-x 4 root root 740 Jan 18 06:12 ..
lrwxrwxrwx 1 root root  10 Jan 17 08:40 usb-CN09GTFM7248753ABEM4A00_Integrated_Webcam_HD-event-if00 -> ../event19
lrwxrwxrwx 1 root root  10 Jan 17 20:21 usb-Corsair_Corsair_K70R_Gaming_Keyboard-event-if01 -> ../event22
lrwxrwxrwx 1 root root  10 Jan 17 20:21 usb-Corsair_Corsair_K70R_Gaming_Keyboard-event-kbd -> ../event21
lrwxrwxrwx 1 root root  10 Jan 17 20:21 usb-Corsair_Corsair_K70R_Gaming_Keyboard-if02-event-kbd -> ../event23
lrwxrwxrwx 1 root root  10 Jan 17 08:40 usb-ELAN_Touchscreen-event-if00 -> ../event15
lrwxrwxrwx 1 root root  10 Jan 17 20:25 usb-Razer_Razer_Mamba-event-mouse -> ../event24
lrwxrwxrwx 1 root root  10 Jan 17 20:25 usb-Razer_Razer_Mamba-if01-event-kbd -> ../event25
lrwxrwxrwx 1 root root   9 Jan 17 20:25 usb-Razer_Razer_Mamba-mouse -> ../mouse3
lrwxrwxrwx 1 root root   9 Jan 17 20:21 usb-Valve_Software_Steam_Controller-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Jan 17 20:21 usb-Valve_Software_Steam_Controller-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Jan 17 08:40 usb-Yubico_Yubikey_4_OTP+U2F+CCID-event-kbd -> ../event6
```

Unfortunately installing Win10 into vmware and passing the mouse through did not show recognition by synapse2 even though I am fairly certain this mouse supported synapse2, I could try the even older legacy driver still available on the razer page if that'd be helpful.

michaelschlies, 2018-01-18:
[image]
[image]
[image]
[image]
[image]
[image]

michaelschlies, 2018-01-18:
So after running with the "legacy" driver 2.01 it looks like this was what Synapse1 would have installed, so I'll let you tell me whether pcaps are worthwhile or not, if the level of effort seems like one that isn't extremely high for you guys I'll go ahead and take a stab at it, otherwise I may just see what the new mambas and deathadders feel like in my hands

z3ntu, 2018-01-28:
Yes, packet captures are definitely needed otherwise we are just shooting in the dark.

michaelschlies, 2018-01-28:
I mean to say, am I the only person with a 2009 mamba and it isn’t worth anyone’s time to map any functionality?

z3ntu, 2018-01-28:
You're probably not 😃 
But anyways if you could give us packet captures then I can tell you how hard it is to add support for the mouse.

cfeckardt, 2018-05-22:
You're not alone. I have one of these mice too. Can provide packet captures/help write driver if needed.

terrycain, 2018-05-23:
@cfeckardt Yes, do both ;)

wschlich, 2019-02-10:
I also have one of these old Mambas (just replaced the dead expanded original battery with a china clone one) and could help out with captures etc. as well :-)

z3ntu, 2019-03-18:
@wschlich Yeah, please do the [packet captures](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) otherwise nothing will progress for this device.

michaelschlies, 2019-03-24:
Funny enough, I actually got rid of my original mamba, bought one of the 2018 Mamba Wireless models only to find it isn't supported either :)

villavic, 2022-10-20:
Here are some pcaps for the 2009 mamba wireless (the dock), on firmware v1.13, with Synapse v2.04. 
<details>
<summary>lsusb -d 1532:000f -v</summary>

```
Bus 009 Device 013: ID 1532:000f Razer USA, Ltd Mamba (Wireless)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x000f Mamba (Wireless)
  bcdDevice            0.01
  iManufacturer           1 Razer 
  iProduct                2 Razer Mamba Charging Dock
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
[... 93 lines cut ...]
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     101
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
</details>

[usb_capture.zip](https://github.com/openrazer/openrazer/files/9830635/usb_capture.zip)

> I don't plan on using this, I grabbed it thinking it didn't work, plugged it in and looked at openrazer support (I'm currently using a Viper), then saw that this issue needed pcaps.

villavic, 2022-10-28:
Here are more actions captured with the mouse plugged in.

<details>
<summary>lsusb -d 1532:000a -v</summary>

```text
Bus 009 Device 026: ID 1532:000a Razer USA, Ltd Mamba (Wired)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x000a Mamba (Wired)
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Mamba
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     101
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
</details>

[usb_capture_wired.tar.xz](https://www.mediafire.com/file/1wfneu6joxd6s9d/usb_capture_wired.tar.xz/file)

z3ntu, 2022-10-29:
@villavic As expected, not using the new style of communication protocol that basically any newer device is using. Feel free to reverse engineer, implement and send a pull request though :D