openrazer/openrazer #507: Is this project still alive?

minecrawler, 2018-01-20:
While there are sporadic bug fixes merged, I have not seen any new device support (and there are many open issues!) over the last months. This leads me to believe, that this project is dead.

Do you plan anything big or am I correct with that assumption? Is there a different driver project, which is more active?

terrycain, 2018-01-20:
Is it alive, ish. Yeah I have nowhere near as much time as i want to devote to this. When I do get time, its spend slowly converting things to not use a kernel space driver, and doing things in userspace with hid... which is going well when I get time. Is there a more active project? I'm not sure, search around github.

We have a telegram chat thats fairly active (add me @terrycain) and I'm more than happy if anyone else wants to add device support or to ask me to figure out how, its not particularly difficult as the majority is already implemented.

z3ntu, 2018-01-21:
Adding to what Terry wrote, I myself don't quite have the skills to add support for completely new devices (at least new protocols) but am mostly fixing various things and being the support person for all of the issues that come in (mostly because of not following the instructions).
I am also involved with other projects that I am doing in my freetime (e.g. mainline kernel for the Fairphone 2 or postmarketOS) because I also have personal interest in them.

About this project: Unfortunately we don't really have many people contributing code to this project which would be the most valuable contribution, so all of the devices that people want to be supported has to be done by ourselves (not counting the occasional device support pull requests).

I am also not aware of any other projects that has the same scope as this one except [razercfg](https://github.com/mbuesch/razer/) but that project mostly only has support for old Razer devices and the device support is minimal compared to this one - and it is more dead than this one 😉 .

enthusiasmus, 2018-01-24:
@terrycain you said:

> I'm more than happy if anyone else wants to add device support or to ask me to figure out how

So... I want to add support for the basilisk mouse. How can I get started to contribute?

Zerophase, 2018-01-24:
I'll be adding more devices whenever I get a new razer product. I also want to update that dbus issue. Sadly, currently I'm busy studying for grad school entrance exams. I should be free to make more frequent contributions in a month or two.

z3ntu, 2018-01-24:
@enthusiasmus You should join our community chat, either via Telegram (https://t.me/joinchat/A1E4RT9TxhnxyetdQCSj7Q) or via Matrix (https://matrix.to/#/#openrazer:matrix.org)

closed 2018-01-28 by z3ntu