openrazer/openrazer #511: Request for Support: Razer Blade 14" 2017 (Kaby Lake)

PDXsailor, 2018-01-24:
Could you please add official support for the Razer Blade 14" 2017 (Kaby Lake)? The Model Number is RZ09-0195, Product Number RZ09-01953E72.

Thanks, Greg

[OpenrazerInfo.txt](https://github.com/openrazer/openrazer/files/1661963/OpenrazerInfo.txt)
[image]
[image]
[image]
[image]
[image]

z3ntu, 2018-01-25:
Please post the output of
`lsusb -d 1532: -v`
and
`ls -lah /dev/input/by-id/`
Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we need packet captures of the usb traffic.

PDXsailor, 2018-01-25:
I already did - see OpenRazerInfo.txt above.

z3ntu, 2018-01-25:
1532:0224 is already supported.

closed 2018-01-28 by z3ntu

PDXsailor, 2018-01-29:
Can the device be added to the readme please?

Razer Blade FHD    1532:0195

Thanks

z3ntu, 2018-01-29:
1532:0195 is not yet supported so no.