openrazer/openrazer #512: Unable to get lock on pidfile

samr28, 2018-01-25:
Hi I'm having trouble getting things to work on my mid 2017 blade stealth. I checked out the troubleshooter and even after adding my user to the plugdev group, `openrazer-daemon -Fv` returns: `Unable to lock on the pidfile.`

I tried: https://github.com/openrazer/openrazer/issues/461 but that didn't solve my issue. I still couldn't get lock on the pidfile. Please let me know what other logs or information would be helpful in diagnosing this problem.

samr28, 2018-01-25:
`$ systemctl --user status openrazer-daemon.service
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Wed 2018-01-24 19:29:15 CST; 13h ago
     Docs: man:openrazer-daemon(8)
  Process: 3012 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=0/SUCCESS)
 Main PID: 3012 (code=exited, status=0/SUCCESS)`

`$ python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[<RazerKeyboard BY1725A10501242>]`

z3ntu, 2018-01-25:
The output with [<RazerKeyboard BY1725A10501242>] says it works ¯\_(ツ)_/¯

z3ntu, 2018-01-25:
And you have to invoke systemctl with `--user` as openrazer-daemon.service is a user unit.

closed 2018-01-25 by z3ntu

samr28, 2018-01-25:
```
$ systemctl --user start openrazer-daemon
Job for openrazer-daemon.service failed because the control process exited with error code.
See "systemctl status openrazer-daemon.service" and "journalctl -xe" for details.
```

z3ntu, 2018-01-25:
pkill -e openrazer
systemctl --user start openrazer-daemon.service

samr28, 2018-01-25:
Ok. It is now active and running but it doesn't seem to be working. Here is the end of the daemon log:

```
2018-01-25 09:13:32 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 168)
2018-01-25 09:13:32 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-01-25 09:13:32 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 166)
2018-01-25 09:13:32 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-01-25 09:13:32 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 163)
2018-01-25 09:13:32 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-01-25 09:13:32 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 161)
2018-01-25 09:13:32 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-01-25 09:13:32 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 158)
```

RazerGenie is sending to the daemon and the daemon is trying to set them but nothing happens on the keyboard. It stays dark :(

z3ntu, 2018-01-25:
Did you turn up the brightness? (the slider might show 100% but move it a bit)

samr28, 2018-01-25:
I moved the slider around and saw those messages in the daemon log. I got the following log messages when attempting to put the brightness on max:

```
2018-01-25 09:19:54 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-01-25 09:19:54 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 242)
2018-01-25 09:19:54 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-01-25 09:19:54 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 245)
2018-01-25 09:19:54 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-01-25 09:19:54 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 247)
2018-01-25 09:19:55 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-01-25 09:19:55 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 250)
2018-01-25 09:19:55 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-01-25 09:19:55 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1725A10501242, 'setBrightness', 252)
```

I also tried changing the effect but it still didn't turn on.

samr28, 2018-01-25:
I did notice this when changing the brightness (ALT + F8/F9):
```
2018-01-25 09:28:00 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 224
2018-01-25 09:28:00 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 224
2018-01-25 09:28:00 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 224
2018-01-25 09:28:01 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 224
2018-01-25 09:28:04 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 224
2018-01-25 09:28:06 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 224
2018-01-25 09:28:06 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 225
2018-01-25 09:28:06 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 225
2018-01-25 09:28:06 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 225
2018-01-25 09:28:07 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 386, in key_action
    key_name = self.EVENT_MAP[key_id]
KeyError: 225
```

samr28, 2018-01-25:
Why was this closed? Is it a mistake on my end? If so, can you please let me know what I did wrong so that I can fix it?

z3ntu, 2018-01-26:
Please join our community chat, either via Telegram (https://t.me/joinchat/A1E4RT9TxhnxyetdQCSj7Q) or via Matrix (https://matrix.to/#/#openrazer:matrix.org). That way we can help you better.