openrazer/openrazer #517: Orbweaver Chroma works only for one keypress

sin0light, 2018-01-30:
Hello !

I'm trying to use my Orbweaver Chroma on Debian 9.0 but I'm stuck right now : 

When I press a key for the first time (after boot or a deamon restart) it works and then nothing (when I press again, any key, nothing happens) whereas the keypresses are detected because they are shown in ~/.local/share/openrazer/logs/razer.log. I managed to having it working but I don't know how (after several plug/unplug, deamon restart, reboot, usb exchange between deathadder and orbweaver, etc...). However lighting is working fine.

I don't understand why, everything is installed as instructed, I've done everything concerning troubleshoot, my deathadder chroma is working fine.

Does anyone have any idea ?
Thanks in advance.

Likutar, 2018-02-12:
Same issue is happening here with my orbweaver chroma the first/second keypress works then nothing

nicodemus144, 2018-02-13:
I'm also experiencing this.

z3ntu, 2018-02-13:
Does the issue also exist when openrazer is not installed?

sin0light, 2018-02-13:
As far as I remember when openrazer was not installed, the orbweaver chroma did not work at all. Any keypress resulted in nothing.

z3ntu, 2018-02-13:
I guess this is a bug in the Linux kernel then...

sin0light, 2018-02-13:
Do you have an idea for fixing it ?

z3ntu, 2018-02-13:
No I have never looked at the kernel side of USB stuff. Maybe file a bug there and see what they say.

omgimdrunk, 2018-02-14:
I had this issue with the Tartarus, I found in the kernel messages that it was one of the GUI frontends that caused the driver to stop responding. Try uninstalling the GUIS and reinstall the drivers, that's what worked for me

potatoattack, 2018-03-02:
I'm having this exact issue with an orbweaver (non-croma), the issue is NOT present with openrazer uninstalled.
As above the inputs are still logged in ~/.local/share/openrazer/logs/razer.log

z3ntu, 2018-03-02:
@potatoattack did you try what @omgimdrunk was suggesting? So uninstall the frontends (polychromatic, RazerGenie, razerCommander - depending which are installed) and reinstall the driver and reboot?

potatoattack, 2018-03-02:
@z3ntu I think that fix was due to the frontends starting the openrazer daemon.
I currently do not have any front ends installed.
Running:
systemctl --user start openrazer-daemon.service
Causes the Orbweaver to stop working after 1 input.

systemctl --user stop openrazer-daemon.service
and it works again.

z3ntu, 2018-03-02:
Yes, we have registered our daemon to autostart when the dbus endpoint is accessed. Thanks for confirming this issue.

z3ntu, 2018-03-02:
Can you try commenting out https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/keyboards.py#L228 (located probably at `/lib/python3*/*packages/openrazer_daemon/hardware/keyboards.py`)

potatoattack, 2018-03-02:
Sure, the issue is not affected when that line is commented out.

potatoattack, 2018-03-03:
Hi, I had a second look at your request and realised that the commented line was in the RazerOrbweaverChroma class. My device is a RazerOrbweaver. I commented out:
[https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/keyboards.py#L192](https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/keyboards.py#L192)
and the problem is fixed.

closed 2018-07-31 by stale[bot]