openrazer/openrazer #519: opensuse tumbleweed fails to run daemon

meop, 2018-02-01:
Hello, I have followed the instructions for Opensuse and running Tumbleweed.. added my user to plugdev and rebooted as well.. is there some unmet dependency still that would cause the error below? Thank you..

mp@linux:~> groups
users plugdev wheel
mp@linux:~> openrazer-daemon -Fv
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 12, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/daemon.py", line 21, in <module>
    gi.require_version('Gdk', '3.0')
  File "/usr/lib64/python3.6/site-packages/gi/__init__.py", line 134, in require_version
    (namespace, version))
ValueError: Namespace Gdk not available for version 3.0

z3ntu, 2018-02-01:
You seem to be missing a dependency on a package that provides `/usr/lib/girepository-1.0/Gdk-3.0.typelib`, that package should be a dependency on the openrazer-daemon package but apparently I missed that.

z3ntu, 2018-02-01:
Apparently it's typelib-1_0-Gtk-3_0 that's missing.

meop, 2018-02-01:
Thank you, that was a fast reply! Adding that package gets the daemon to run, but the kernel module razermouse does not load.. is that also packaging related, or something else?

mp@linux:~> sudo journalctl -b | grep -i razer_mount
Jan 31 23:14:56 linux razer_mount[809]: Driver razermouse
Jan 31 23:14:56 linux razer_mount[811]: Driver razermouse
Jan 31 23:14:56 linux razer_mount[812]: Driver razermouse
Jan 31 23:14:56 linux razer_mount[813]: Device_ID 0003:1532:005C.0004
Jan 31 23:14:56 linux razer_mount[815]: Device_ID 0003:1532:005C.0005
Jan 31 23:14:56 linux razer_mount[817]: Device_ID 0003:1532:005C.0003
Jan 31 23:14:56 linux razer_mount[819]: Modprobing razermouse
Jan 31 23:14:56 linux razer_mount[818]: Modprobing razermouse
Jan 31 23:14:56 linux systemd-udevd[474]: Process 'razer_mount razermouse 0003:1532:005C.0005' failed with exit code 1.
Jan 31 23:14:56 linux razer_mount[823]: Modprobing razermouse
Jan 31 23:14:56 linux systemd-udevd[512]: Process 'razer_mount razermouse 0003:1532:005C.0004' failed with exit code 1.
Jan 31 23:14:56 linux systemd-udevd[502]: Process 'razer_mount razermouse 0003:1532:005C.0003' failed with exit code 1.
mp@linux:~> lsusb
Bus 002 Device 002: ID 05e3:0612 Genesys Logic, Inc. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1532:005c Razer USA, Ltd 
Bus 001 Device 003: ID 04d9:0169 Holtek Semiconductor, Inc. 
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
mp@linux:~> lsmod|grep razer
mp@linux:~>

z3ntu, 2018-02-01:
What happens when you run `sudo dkms install openrazer-driver/1.0.0` ?

meop, 2018-02-01:
```
mp@linux:~> sudo dkms install openrazer-driver/1.0.0
Module openrazer-driver/1.0.0 already installed on kernel 4.14.15-1-default/x86_64
```

meop, 2018-02-01:
That worked. And after a reboot, razermouse did too. Looks all fixed. Thank you.

z3ntu, 2018-02-01:
Fixed by https://github.com/openrazer/OBS-packaging/commit/154b4b21ceba5d27279d2e93c883467f2ddccec9

Charadon, 2020-04-28:
I don't think this packaging issue was resolved, as my friend who uses opensuse got this exact same error, and was missing the exact same dependency: typelib-1_0-Gtk-3_0 
I think this package needs to be precisely named for it to be listed as a dependency.

z3ntu, 2020-04-28:
Hm `typelib(Gdk)` seems to he provided by `typelib-1_0-Gtk-2`, `typelib-1_0-Gtk-3` and `typelib-1_0-Gtk-4` so I'm guessing a different version than the 3.0 one wrong version from that was installed.

closed 2020-05-01 by z3ntu