openrazer/openrazer #520: Razer Blade Stealth (Late 2017) isn't recognized by driver

Rasie1, 2018-02-02:
Just got the RBS Late 2017 and I'm really happy with it (however, still didn't fix the screen flicker and touchpad random clicking), but the openrazer drivers don't work.

I'm using manjaro (`4.14.15-1-MANJARO`) and driver doesn't see the device. I tried `openrazer-meta` and `openrazer-meta-git` from AUR, both give me same results. I added myself to `plugdev` group as installation asked and I rebooted the device.

Daemon output with `-Fv`
```
Starting daemon.
2018-02-02 16:21:01 | razer                          | INFO     | Initialising Daemon (v2.2.0). Pid: 4229
2018-02-02 16:21:01 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerMambaTE ---------------------------- (1532:0046)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerNagaChroma ------------------------- (1532:0053)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaverChroma -------------------- (1532:0207)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2018-02-02 16:21:01 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
[... cut ...]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
$ lsmod | grep razer
```

Here is what I have installed currently
```
[rasiel@rasiel ~]$ pacaur -a -Qs | grep razer
local/openrazer-daemon-git 2.2.0.r11.gc557c6a-1
local/openrazer-driver-dkms-git 2.2.0.r11.gc557c6a-1
local/openrazer-meta-git 2.2.0.r11.gc557c6a-1
    Meta package for installing all required openrazer packages. (Git version)
local/python-openrazer-git 2.2.0.r11.gc557c6a-1
```

If there is any way I can help, please ask me.

z3ntu, 2018-02-02:
The output of `sudo dkms install openrazer-driver/1.0.0` and of `sudo modprobe razerkbd && dmesg | tail` please

Rasie1, 2018-02-02:
```
[rasiel@rasiel ~]$ sudo dkms install openrazer-driver/1.0.0
[Module openrazer-driver/1.0.0 already installed on kernel 4.14.15-1-MANJARO/x86_64
[rasiel@rasiel ~]$ sudo modprobe razerkbd && dmesg | tail
modprobe: ERROR: could not insert 'razerkbd': Exec format error
```

Installation after removal

```
[rasiel@rasiel ~]$ sudo dkms install openrazer-driver/1.0.0

Creating symlink /var/lib/dkms/openrazer-driver/1.0.0/source ->
                 /usr/src/openrazer-driver-1.0.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
KERNELDIR=/lib/modules/4.14.15-1-MANJARO/build make driver.....
cleaning build area...
Kernel cleanup unnecessary for this kernel.  Skipping...

DKMS: build completed.

razerkbd.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.14.15-1-MANJARO/kernel/drivers/hid/

razermouse.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.14.15-1-MANJARO/kernel/drivers/hid/

razerfirefly.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.14.15-1-MANJARO/kernel/drivers/hid/

razerkraken.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.14.15-1-MANJARO/kernel/drivers/hid/

razermug.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.14.15-1-MANJARO/kernel/drivers/hid/

razercore.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/4.14.15-1-MANJARO/kernel/drivers/hid/

depmod.....

DKMS: install completed.
[rasiel@rasiel ~]$ sudo modprobe razerkbd && dmesg | tail
modprobe: ERROR: could not insert 'razerkbd': Exec format error
```

z3ntu, 2018-02-02:
Sorry, `sudo modprobe razerkbd; dmesg | tail` please

Rasie1, 2018-02-02:
```
modprobe: ERROR: could not insert 'razerkbd': Exec format error
[26482.270039] ath: EEPROM regdomain: 0x8324
[26482.270043] ath: EEPROM indicates we should expect a country code
[26482.270046] ath: doing EEPROM country->regdmn map search
[26482.270049] ath: country maps to regdmn code: 0x3
[26482.270052] ath: Country alpha2 being used: UA
[26482.270053] ath: Regpair used: 0x3
[26482.270057] ath: regdomain 0x8324 dynamically updated by country IE
[27154.057111] razerkbd: version magic '4.14.15-1-MANJARO SMP preempt mod_unload modversions retpoline ' should be '4.14.15-1-MANJARO SMP preempt mod_unload modversions '
[27284.543177] razerkbd: version magic '4.14.15-1-MANJARO SMP preempt mod_unload modversions retpoline ' should be '4.14.15-1-MANJARO SMP preempt mod_unload modversions '
[27468.532079] razerkbd: version magic '4.14.15-1-MANJARO SMP preempt mod_unload modversions retpoline ' should be '4.14.15-1-MANJARO SMP preempt mod_unload modversions '
```

z3ntu, 2018-02-02:
This seems to be a Manjaro issue. Please complain there.
https://forum.manjaro.org/t/manjaro-4-14-15-version-magic-doesnt-match-prevents-module-from-loading/39318

closed 2018-02-02 by z3ntu

Rasie1, 2018-02-02:
Thanks! Also, by the way, when I run it again, I get slightly different message
```
[rasiel@rasiel ~]$ sudo modprobe razerkbd; dmesg | tail
[26482.270052] ath: Country alpha2 being used: UA
[26482.270053] ath: Regpair used: 0x3
[26482.270057] ath: regdomain 0x8324 dynamically updated by country IE
[27154.057111] razerkbd: version magic '4.14.15-1-MANJARO SMP preempt mod_unload modversions retpoline ' should be '4.14.15-1-MANJARO SMP preempt mod_unload modversions '
[27284.543177] razerkbd: version magic '4.14.15-1-MANJARO SMP preempt mod_unload modversions retpoline ' should be '4.14.15-1-MANJARO SMP preempt mod_unload modversions '
[27468.532079] razerkbd: version magic '4.14.15-1-MANJARO SMP preempt mod_unload modversions retpoline ' should be '4.14.15-1-MANJARO SMP preempt mod_unload modversions '
[27544.012924] [drm:intel_cpu_fifo_underrun_irq_handler [i915]] *ERROR* CPU pipe A FIFO underrun
[27780.493302] razerkbd: bad vermagic: kernel tainted.
[27780.493305] Disabling lock debugging due to kernel taint
[27780.493312] razerkbd: loading out-of-tree module taints kernel.
```

z3ntu, 2018-02-02:
Yeah the kernel complains that you force something down its throat that it doesn't like.

Rasie1, 2018-02-02:
Yes, I tried `--force-vermagic`, but looks like it didn't help and I'm going to investigate fixes they talk about in the thread

Rasie1, 2018-02-20:
I succeeded when I installed kernel 4.15